mjansen4857/pathplanner #5: Delete Points

jasondaming, 2019-01-22:
I have right clicked to add a point how can I delete it?  When I right click a menu comes up to input the point information but there is no cancel button

jasondaming, 2019-01-22:
I now see in the readme that I can control right click the point.  A shortcuts legend might be helpful.  

Also units.  I can guess what units are coming out but I am not sure, inches or feet // degrees or radians?

mjansen4857, 2019-01-22:
I agree. I'll work on adding a list of shortcuts to the readme. As for the units, distances are either feet or meters depending on which unit system is selected. Angles are in degrees. I'll try to make this more clear as well.

mjansen4857, 2019-01-22:
The readme has been updated and I pushed a commit that adds undo and redo shortcuts. Working on releasing now.

closed 2019-01-22 by mjansen4857