mjansen4857/pathplanner #6: Add time in ms to output

team3572, 2019-01-24:
Please add an option for time, in ms, for the output.
The motion profile code I'm working off from seem to all want time in ms as a third parameter
example:
0,0,20
0.0011,0.0543,20
0.0042,0.1579,20
0.0095,0.2624,20
...
So my specifier would be p,v,t
and time is "Time Step" * 100

Thanks.

team3572, 2019-01-25:
Sorry, I should have been more clear. The time I needed output to the CSV file was not total elapsed time for the path up to that row but simply the exact value from "Settings->Time Step". So given a Temp Step of ".02" my output for "p,v,t" would look like this... (of course I'd prefer the output to be Time Step * 100)
```
0,0,.2
0.0016,0.0797,.2
0.0058,0.2136,.2
0.0129,0.3523,.2
0.0226,0.4913,.2
0.0353,0.6308,.2
0.0506,0.7706,.2
0.0688,0.91,.2
0.0898,1.0502,.2
0.1136,1.1902,.2
```

Thanks for all your effort and help.

mjansen4857, 2019-01-25:
Ok I understand now. I left in the time elapsed option and added a time step option as well. Capital S is time step in seconds, lowercase s is time step in ms.