mjansen4857/pathplanner #7: Deploying with Time Output Format

hamster1147, 2019-01-25:
Deploying to robot with Output Format 't' throws an error message "Invalid Output Format".

Also the documentation doesn't reflect the addition of flag 't'.

hamster1147, 2019-01-25:
I just realized that I was using the recent release and not testing the source. When I tested this with the most recent relase, I was able to generate the CSV file locally successfully, but when using the deploy to robot it gave me the "Invalid Output Format" error. Removing 't' from Output Format works and it successfully deploys to the robot directly.

mjansen4857, 2019-01-25:
Is your path location in settings `/home/lvuser/deploy/paths`? I'll play around with deploying when I get to the robot on Saturday. Not sure what was causing the error I was getting.

hamster1147, 2019-01-25:
No, I'm using our path "/home/lvuser/AutonomousTrajectories/", but it works when I don't use 't' as an Output Format.

This tool is great by the way. I'm really enjoying your work.

hamster1147, 2019-01-25:
Just tested the new build. Deploying to robot works with 't' as an Output Format.

closed 2019-01-25 by hamster1147