mjansen4857/pathplanner #9: Deploying to robot ignores Split Path switch

hamster1147, 2019-01-25:
Currently deploying to the robot sends both a right and left path no matter which position the split path switch is set to. When generating the path locally, only a source path is generated with the split path turned off, which I assume is working as intended. When deploying to robot, the program seems to ignore the split path switch.

mjansen4857, 2019-01-25:
Hmm. I thought I made it work for deploying as well. I'll take a look.

mjansen4857, 2019-01-25:
This should be fixed now. I don't have access to a robot to test it right now so let me know if it still doesn't work.

hamster1147, 2019-01-25:
Tested it on our robot and it works as intended.

Thanks for fixing these issues so fast. I was able to get our robot to follow the generated paths pretty closely, your tool is so powerful. Thank you for your work.

hamster1147, 2019-01-25:
So I tested it and it did send the center only file, but it seems the connection to the robot's file system is finicky. I haven't been able to debug it yet, but sometimes I can connect over sftp but not be able to deploy to robot from pathplanner. A lot of the time, but not all of the time, it fails to connect.

mjansen4857, 2019-01-25:
When you connect to the robot is it the only network connection on your PC, or are you connected to the internet and the robot? I'm not sure what else could cause it to fail if you're connected to the robot. The library I use to deploy uses SFTP so if you can connect with SFTP manually it should be working. I'll have access to some RIOs tomorrow so I will play around with it then.

There is a log file inside of a .PathPlanner folder in your user folder. For windows this would be:
 `C:\Users\<Your User>\.PathPlanner\log.txt`
Errors while connecting should be logged in there. Can you check if there's any connection errors?

hamster1147, 2019-01-25:
Came back to the shop again, it worked. Don't know why it didn't work before.

mjansen4857, 2019-01-26:
Interesting. If it fails again see if there are any errors in the log and let me know.

hamster1147, 2019-01-26:
Roger

hamster1147, 2019-02-03:
It seems to be doing it again. When I generate the path locally, I only get the one file. When I deploy I get both the left and right file but not the center.

mjansen4857, 2019-02-04:
I'm looking at the code and it seems like it should be working correctly. I will be able to test it on a robot later today and see if I can reproduce the issue.

hamster1147, 2019-02-05:
Built the most current version. It is gone.

closed 2019-02-05 by hamster1147