mjansen4857/pathplanner #13: [request] End point velocity

hamster1147, 2019-02-11:
If the last point is set to a certain velocity using the velocity override, it would be helpful if the path ended in the position at that velocity while following all other rules.

mjansen4857, 2019-02-12:
The velocity override only changes the maximum velocity at that point as well as +/- 1/4th of the way to the next point. Ending a path while still moving would remove the accuracy of motion profiling in my opinion, and this would break the way we plan to use the feature ourselves. Therefore, I don't plan on adding this into PathPlanner, however, this is easy to implement yourself if you would like to do this. All you have to do is replace this line:

https://github.com/mjansen4857/PathPlanner/blob/a539c59796f4e1f1d1ac5b522ed2672b6d5bab82/path_generator.js#L334

with this:

```
if(this.velocities[this.velocities.length - 1] == this.maxVel) {
	p[p.length - 1].vel = 0;
}
```
then build the app from source.

That line sets the velocity at the last point to zero so that the robot will stop, adding the if statement will prevent this from happening if the velocity is overridden for the last point, causing the path to end at that velocity.

closed 2019-02-12 by mjansen4857

lidor51, 2019-12-15:
If you set as default the end velocity to 0, how would it break this feature and decrease the accuracy?

Finishing in non-zero velocity could be beneficial in cases you want to jump smoothly from motion profiling to vision control.

mjansen4857, 2019-12-24:
I played around with this a while ago and ended up adding it. It will be in the next release.

hamster1147, 2020-01-27:
That's good to hear. When I for this last year, that was the main reason why. We ended up changing a bit of the velocity profile generation and got really good results. Your tool was a huge help to me last year. [Here's](https://www.youtube.com/watch?v=J2LB8LqO1Z0) a video of our main autonomous mode that we used. Everywhere you don't see the limelight used during auto, the robot is following paths generated by a modified version of your tool. We used it a bunch for several auto paths. Thanks for your work.

mjansen4857, 2020-01-27:
I'm glad it worked well for you! The auto looks great. Our team mostly did driver control during sandstorm last year so I'm excited to see what they do this year.