mjansen4857/pathplanner #25: Always defaults to finishing a path at max velocity instead of stopping.

bob80333, 2020-01-25:
It looks like this is related to the ability to override final point velocity, but there doesn't appear to be a way to make it default to stopping.  The 'generate path' section of the readme mentions it, but there is no option in the generate GUI.

bob80333, 2020-01-25:
Additionally, when overriding the max velocity of the last point, it appears the minimum velocity is 1, so your robot cannot come to a stop.  

When overriding the velocity of the last point, the robot slows to the max velocity far before it actually finishes moving, making the path much slower than it should be.  In a test where I set the start and end points to (10, 10) and (20, 10) both with angle 0 and overrode the last point to max velocity 1 with everything else at default values, the trajectory slowed to a velocity of 1 by line/timestep 215 in the CSV, but the trajectory only finished at line/timestep 551,  more than 300 timesteps later!

mjansen4857, 2020-01-25:
Someone else emailed me about this issue and I realized I need to update it to make it more clear. In order to have the robot stop at the last point you need to delete the number from the velocity endpoint box and hit confirm. That box gets auto populated with the max velocity as a placeholder even if it will follow the default behavior of stopping. I will change it to not auto populate it at all.

It is intentional to have a minimum value of one so that you can't enter 0 and have a path that takes infinite time to complete. The velocity override is applied to 25% (or 50% I can't remember) of the distance between the overridden point and the points before and after it. This is to give the velocity override more utility since it will affect a larger area and not just a singular point. For example, limiting the velocity while the robot travels along a curve.

bob80333, 2020-01-25:
Thanks for the clarification.  The behavior for overridden velocities makes sense, but when trying to make the last point stop movement they were the opposite of what I wanted.  Perhaps a GUI switch in the generation panel to turn on or off stopping at the last point in a path would be clearer.

closed 2020-01-25 by bob80333