mjansen4857/pathplanner #27: Absolute Winding and Heading.

lidor51, 2020-02-11:
Although it worked in the summer, it doesn't export the absolute winding nor absolute heading.
I'm trying to export W,H but it always export it as a relative winding/heading.

Also when you fix it, please remember that Absolute Heading & Winding in reverse mode is +180 degrees from forward mode.
[ if theta is the **initial** heading of forward -> **initial**_reverse_absolute_heading = (theta+180)mod360 ]

mjansen4857, 2020-02-17:
Capital W/H is for relative winding and heading. Absolute heading and winding should work for lowercase w and h.

lidor51, 2020-02-17:
I tried both ways, but it didn't work..

mjansen4857, 2020-02-17:
Can you give me an example of a path you tried that didn't work, as well as what the output was? The path I tried seemed to work fine.

mjansen4857, 2021-01-14:
Going to close this due to inactivity. Let me know if this is still an issue.

closed 2021-01-14 by mjansen4857