mjansen4857/pathplanner #29: Path does not end with zero velocity

wezelball, 2020-02-20:
I'm experimenting with a single short path where the robot ends up 10 feet in the x ad y direction, at an angle of 0 degrees. The path velocity never ends up a 0 ft/sec, even when I override it.  The override seems to have no effect.

mjansen4857, 2020-02-20:
Do you mean that the end velocities look similar to this?

```
0.2497
0.1997
0.1475
0.0967
0.0367
```

The final point will rarely if ever, have an end velocity of zero. This is due to how the values are calculated during generation. The velocity of the last point is small enough that it should have a negligible impact on where the robot stops. 

You also mentioned the override, so if your end velocity is >= 1 then its because you have a velocity override on that point. You can't enter an override of zero because the override affects more than a single point in time, thus causing a path with infinite length because it will never reach the end. If you want the robot to stop at the last point you need to remove the velocity override for that point.

Do either of those describe what's happening or are you experiencing a different issue?

wezelball, 2020-02-21:
Okay, I tried a few different paths and got the near-zero numbers.  I see what you mean.  I'll close the issue and thanks for responding and helping me out.

closed 2020-02-21 by wezelball