mjansen4857/pathplanner #42: Maximum velocity, acceleration constraints not applied to left/right paths

LeeClarkSystechOne, 2021-01-25:
Paths for the 2021 at home challenge are carefully constructed to 'look nice' (wheel path display is visibly smooth).  The 't, v, vr, vl' columns in the generated paths are plotted in Excel. The velocity graph for the robot center (v) is fully constrained by max vel/accel. The plots for vr & vl show discontinuities, slopes exceeding max accel, and velocities exceeding max velocity. Perhaps there should be separate configuration parameters for max wheel vel/accel? Would a menu option to adjust the splines to remove minor l/r discontinuities be appropriate?

Sample path, output, and plot attached.
[Slalom Path.zip](https://github.com/mjansen4857/PathPlanner/files/5867697/Slalom.Path.zip)

mjansen4857, 2021-03-09:
Sorry for the super late response on this, I've been pretty busy and haven't really had any time to dedicate to this project.

The problem of maximums not being respected, discontinuities, etc. when splitting a path has been a known issue to me for a while now but I never dealt with it because through my team's usage we never had any real issues with it while path following. I don't usually recommend splitting the path anymore anyways since we seem to get better results using a center path for position/velocity and an IMU for heading.

If you've found that these issues cause noticeable problems with your path following I would be happy to take a look at a potential fix but I currently think that the effects of this are fairly insignificant.

mjansen4857, 2021-11-06:
Closing this due to inactivity and since splitting paths is no longer a part of the generation algorithm

closed 2021-11-06 by mjansen4857