mjansen4857/pathplanner #55: Paths names

lidor51, 2021-12-25:
I noticed that if you press "Shift" when editing path name, is treat as you would press "CapsLock".

mjansen4857, 2021-12-29:
I'm unable to reproduce this. Could you go more into depth on steps to reproduce or provide a screen recording of it happening or something? Have you ruled out the possibility of it being a hardware issue with your keyboard?

lidor51, 2021-12-29:
Pressed shift to insert a capital letter, and after release, the following letter were upper case as well. The only way I can handle it, was go to another program, change to low case again, then go to PathPlanner.

mjansen4857, 2021-12-29:
I can't reproduce this no matter what I do. If it is a bug it's likely a flutter bug which I can't really do anything about. I've upgraded the flutter version recently so I guess just wait and see if it's fixed whenever a new PathPlanner update comes out.

mjansen4857, 2022-03-19:
Is this still an issue?

closed 2022-04-16 by mjansen4857