mjansen4857/pathplanner #57: Velocity Override doesn't work

lidor51, 2021-12-25:
In both Holonomic & Differential Drive, the Velocity Override I wrote in a Waypoint doesn't get saved.

mjansen4857, 2021-12-29:
Fixed the velocity override issue in c7db3b9, however this will also require a PathPlannerLib change before it works on robot. I can't reproduce the not saving issue. Are you hitting the enter key after entering a value?

lidor51, 2021-12-29:
Yes, I pressed enter for sure.

mjansen4857, 2021-12-29:
I fixed an issue with the velocity override text boxes not showing the correct value. Maybe that is the issue you are talking about?

amquake, 2022-01-20:
When setting the velocity override on the last waypoint, it doesn't seem to have an effect. Whatever value is specified is just ignored.
https://user-images.githubusercontent.com/7953350/150341161-9978cf31-8678-4660-a156-c0afc6090087.mp4
I figured out you could empty the text box and press enter to reset it to the null, which returned it to the default behavior.
https://user-images.githubusercontent.com/7953350/150341169-c90bd486-a42f-468e-af06-3e4485a89f07.mp4

mjansen4857, 2022-01-20:
This is technically functioning as expected. A velocity override of 0 is not allowed since it would create an infinite loop in the generation code. I will change it to treat a 0 as a null value to prevent this from happening though.

mjansen4857, 2022-03-19:
Fixed in [9a41c25](https://github.com/mjansen4857/pathplanner/commit/9a41c252007ee6cfcf8be0cb3fd7c1baff76208f)

I believe the issues with the velocity override box have been fixed now. If there's still issues after next release then reopen this.

closed 2022-03-19 by mjansen4857