mjansen4857/pathplanner #61: glibc 2.34 dependency

viggy96, 2022-01-19:
Is it possible for you to downgrade the version of glibc required? It appears that you're requiring version 2.34, however that version isn't available, even on Arch Linux. Arch/Manjaro does have 2.33 available.

mjansen4857, 2022-01-20:
Flutter automatically makes these dependency requirements when building. I believe it bases the requirements on what version your distro has. If you build manually this should resolve the issue for you. However, I noticed that I have been building with Ubuntu 21.10, I just updated the 2022.1.0 release with a build from Ubuntu 20.04. Can you try this build and see if that fixes your problem since it has older dependencies?

viggy96, 2022-01-20:
Yup, that updated build worked for me. Thanks for your work, I look forward to using PathPlanner this season.

closed 2022-01-20 by viggy96