mjansen4857/pathplanner #63: Differential Drive Paths Inaccuracy (vs PathWeaver)

carelesshippo, 2022-01-25:
Hi. So I was having a bit of trouble with path planner accuracy when turning on a real robot so I created a simple simulation and created very similar paths with the same constraints in each (3 m/s max velocity, 3 m/s^2 max acceleration). The results are below. The repo for the code I did to test this is with is [here](https://github.com/carelesshippo/PathPlannerTest). Thanks for your help in advance. No rush, I imagine you are very busy with your own build season. 
## PathWeaver
[image]
## PathPlanner
[image]

mjansen4857, 2022-01-26:
Have you tried tuning your b and zeta constants? I'm not sure why your controller would work differently for the different paths.

carelesshippo, 2022-01-26:
> Have you tried tuning your b and zeta constants? I'm not sure why your controller would work differently for the different paths.

Thanks for your help. Setting it to around 10 seemed to fix it.

closed 2022-01-26 by carelesshippo

mjansen4857, 2022-01-26:
Interesting. Do these values work fine for different paths as well?

mjansen4857, 2022-01-26:
If it works it works I guess. I'm not too familiar with using the Ramsete controller so I'm not sure what else could cause it to be that different.