mjansen4857/pathplanner #66: The error of the rotation in the PathPlanner

mendax1234, 2022-01-30:
First of all, thanks for developing such an amazing tool. From my perspective , this is the best swerve auto path planner in the FRC now, with the easiest way to use. And with many people's help , l have now tested the path with rotation successfully on my 2 wheel swerve(locate in a diagonal line). However , l still find some bug with this tool.

The bug lies in the rotation of the waypoint. As we all know , in FRC ,we use a field cordinate system and theta is in the counter-clockwise direction. Just like the graph below.

[image]

However , in the pathplanner tool. It seems that the x and y axis is correct , but the theta direction is incorrect. For instance, when l set the rotation to 90 degree in the GUI. But in the real situation , my swerve will rotate to -90 degree according to its gyro(l have used my gyro to check my robot angle and it is right.)

[image]

From the GUI , the rotation is right . So , l think the bug lies in the lib . Hope that you can fix the bug and release a new version of lib soon!

bruingineer, 2022-01-30:
I agree this a great tool for swerve drive path planning. 

The gyro should be reporting an increasing angle as the robot turns counter clockwise. If using a navX, you need to negate the reported angle. Have you verified this already? 

When sending the gyro value to Shuffleboard or smartdashboard, it should also be negated to match the gyro value of what odometry and other WPI classes expect.

mjansen4857, 2022-01-30:
I agree with @bruingineer 

If you display the path on a Field2D SmartDashboard widget it will rotate as expected, so I don't think there's anything wrong with the generation of the rotation. Try negating the value of your gyro everywhere it is used and see if it works.

mendax1234, 2022-01-31:
> If using a navX, you need to negate the reported angle. Have you verified this already?

I use navX as my gyro and l have verified this .

> When sending the gyro value to Shuffleboard or smartdashboard, it should also be negated to match the gyro value of what odometry and other WPI classes expect

Yeah , l have created a method to get this negative value and show it in the shuffleboard.

> Does your robot physically rotate in the direction expected based on PathPlanner trajectory?

Probably not , in the trahectory , it should rotate to +90 degree , but actually , it rotates to -90 degree. Besides , the 0 degree in the rotation leads to the +180 degree(or -180) rotation physically. All of the rotation in the pathplanner is 180 degree bigger than the physical rotation.

And my method to get the robot rotation is below:
```
  public Rotation2d GetGyroRotation2d(){
    return new Rotation2d(ahrs.getYaw() * -1 / 180 * Math.PI);
  }
```

bruingineer, 2022-01-31:
Try using `getAngle()` instead of `getYaw()`. `getYaw()` returns a range of (-180 to 180), while `getAngle()` returns the total accumulated angle, which should play nicer with the rotation PIDController. 

try this as the function to get the Rotation2d:
```
public Rotation2d getGyroRotation2d() {
    return Rotation2d.fromDegrees(-m_gyro.getAngle());
}
```

mendax1234, 2022-01-31:
> which should play nicer with the rotation PIDController.
l have tested and it has no difference . 

Besides , l am very sure that there is nothing wrong with my gyro. And l want to see if anybody has the same issue.

mjansen4857, 2022-01-31:
You definitely want the range to be between -180 and 180 because that is the range that the holonomic rotation is in as well. You need to make sure your theta controller has continuous input enabled. 

Try adding 180 degrees to your gyro angle. You will need to make sure that the angle stays in between -180 and 180 with something like this:
``` Java
double angle = -ahrs.getYaw() + 180
if(angle > 180){
    angle -= 360;
} else if(angle < -180){
    angle += 360;
}

return Rotation2d.fromDegrees(angle);
```

bruingineer, 2022-01-31:
Our swerve code is currently using `getAngle()`. I'll verify tonight that it can track a trajectory with more than a full rotation.

@mjansen4857 brings up a good point about setting up the theta controller:
> You need to make sure your theta controller has continuous input enabled
```
thetaController.enableContinuousInput(-Math.PI, Math.PI);
```

Double check that you're using the same module order in both 
`SwerveDriveKinematics(...)` 
and when indexing the array returned from 
`var swerveModuleStates = m_kinematics.toSwerveModuleStates(chassisSpeed, centerOfRotationMeters);`

@mendax1234 Do you have your code posted to github?

mendax1234, 2022-02-01:
> You need to make sure your theta controller has continuous input enabled.

Yes , l do have a continuous input for my ProfiledPIDController.

mendax1234, 2022-02-01:
> Double check that you're using the same module order in both

Yeah , l have double checked it recently and it is correct.

> @mendax1234 Do you have your code posted to github?

Not yet, l decide to make it fully open-source at the end of the season. But let me know which part you want to look at , l will give you.

mendax1234, 2022-02-07:
l think this issue do exist and should be fixed . @bruingineer @mjansen4857

bruingineer, 2022-02-07:
how are you initializing the robot start pose before running the trajectory? This might be related.

My team recently ran into an issue when the start heading and rotation differed and we only used the initial pose to reset odometry.
```resetOdometry(new Pose2d(state.poseMeters.getTranslation(), state.holonomicRotation))```

 I believed this was fixed when we used only the translation of the inital pose combined with the holonomic rotation from PathPlanner to reset our odometry.
```resetOdometry(new Pose2d(state.poseMeters.getTranslation(), state.holonomicRotation))```

mendax1234, 2022-02-07:
> resetOdometry(initialState.poseMeters)

What is your initialState, we use *trajectory.getInitialPose()* to reset the odometry.

bruingineer, 2022-02-07:
> we use trajectory.getInitialPose() to reset the odometry.

Instead of `getInitialPose()`, use `getInitialState()` to get the initial `PathPlannerState` of the trajectory, like below:
```
// get initial state from the trajectory
private PathPlannerState getInitialState(PathPlannerTrajectory traj) {
    return traj.getInitialState();
}
```

and then, when resetting odometry, use a new `Pose2d` created with the X-Y translation and the holonomicRotation of the initial state, as below:
```
// reset odometry of swerve drive, 
// passing in a new pose2d with translation of initial pose 
// and the holonomicRotation provided by PathPlanner
resetOdometry(new Pose2d(initialState.poseMeters.getTranslation(), initialState.holonomicRotation))
```

Just for reference, the resetOdometetry function based on WPI and other examples:
```
public void resetOdometry(Pose2d pose) {
    m_odometry.resetPosition(pose, getGyroRotation2d());
}
```

mendax1234, 2022-02-07:
l will try it a week later since l am too busy on my school's homework now. Sorry for that! :upside_down_face:

l think that it may be the solution to this issue but l am not sure.

mjansen4857, 2022-03-19:
Going to close this due to inactivity. Feel free to re-open if you still need help.

closed 2022-03-19 by mjansen4857

mendax1234, 2022-03-19:
Yeah, l have solved this problem and sorry for forgetting to close it!

mendax1234, 2022-03-19:
Besides , when will the latest version be released?

mjansen4857, 2022-03-19:
Going to build and submit everything tomorrow probably. It will still need to go through App Store reviews and what not so a couple days probably.