mjansen4857/pathplanner #67: Can't Save files with ';'

barel13, 2022-02-03:
My team has tried to save files with ':' in their path's name, and suddenly all of the paths disappeared and only one file appeared with the characters before the colon (and without the path extension)

bruingineer, 2022-02-03:
Aside from the library possibly deleting other paths, a `:` is not allowed in a filename of many filesystems. See the decent comparison table here: [https://en.m.wikipedia.org/wiki/Filename#Comparison_of_filename_limitations](https://en.m.wikipedia.org/wiki/Filename#Comparison_of_filename_limitations)

mjansen4857, 2022-03-19:
Fixed in [2298747](https://github.com/mjansen4857/pathplanner/commit/22987471dfd82d69d2f04adeab1ead52cba1ca77)

This will prevent you from typing invalid characters in the text field

closed 2022-03-19 by mjansen4857