mjansen4857/pathplanner #69: Problem moving directly sideways

kinahawi, 2022-02-25:
I've created a simple path to move the bot sideways along the X-Axis. (see picture). When I use the visualization in the path planner tool, the bot moves as expected, but not so on the real bot or the WPILib simulation. As best as I can figure out, the initial pose is returning an angle of zero while the plan has an angle of 90. Any assistance you can provide ASAP is greatly appreciated.

Here is the contents of the path file:
{"waypoints":[{"anchorPoint":{"x":1.0,"y":4.0},"prevControl":null,"nextControl":{"x":1.6939894430884495,"y":4.0},"holonomicAngle":90.0,"isReversal":false,"velOverride":null,"isLocked":false},{"anchorPoint":{"x":3.0,"y":4.0},"prevControl":{"x":2.39241431435475,"y":4.0},"nextControl":null,"holonomicAngle":90.0,"isReversal":false,"velOverride":null,"isLocked":false}]}

[image]

burakdemirelli, 2022-02-26:
Facing the same issue here

kinahawi, 2022-02-27:
I think we figured out what's going on. When we use the getInitialPose function on the trajectory to reset the odometry in our drive, it is not populated with the heading of our robot at the starting point of the plan. We did the following as a work around and things are good now. It grabs the first state and uses the holonomic rotation as part of the starting pose.

        Pose2d temp = trajectory.getInitialPose();
        PathPlannerState s = (PathPlannerState) trajectory.getStates().get(0) ;
        Pose2d temp2 = new Pose2d( temp.getTranslation(), s.holonomicRotation) ;
        m_robotDrive.resetOdometry(temp2);

mjansen4857, 2022-03-19:
This is the correct way to reset odometry. You will always need to get the rotation separately because the rotation component of the Pose2d object needs to be heading or it breaks a lot of things, even on holonomic drive trains. For example, HolonomicDriveController uses the heading to calculate the X and Y velocity feed forwards from a single linear velocity.

closed 2022-03-19 by mjansen4857