mjansen4857/pathplanner #70: 'loadPath' method without velocity or acceleration arguments

viggy96, 2022-03-04:
It would be nice if there was a `loadPath()` method in the PathPlannerLib that didn't need to take in velocity and acceleration as arguments, and instead read them from the path file produced by PathPlanner.

WarrenReynolds, 2022-03-24:
I agree with this request, should just need an overload constructor that only takes the path name and an optional reverse bool, that way if you want to override whats in the json file, then just supply the values, otherwise use the values in the Json file.

WarrenReynolds, 2022-03-24:
This issue should really be in pathplannerlib repo.  To add to this the reversed flag in the Json file is ignored as well.  Maybe a constructor that only takes a filename, then it uses internal Json values for max Vel, Max Acc and reversed.

mjansen4857, 2022-03-24:
I'm a bit hesitant on this at the moment. These values are not guaranteed to be in path files and should not really be considered part of a path's definition like waypoints are. These values are only in the path files so that their corresponding values in the GUI persist. If these values are not present in the path file then this method has no choice but to throw an exception since assuming robot information like vel/acc with default values is not a great idea. Throwing exceptions like that is something I really want to avoid along with the confusion it can cause. For example, if someone loads a path with this method that has the values in it, the method will work fine, but then if they change it to load a different path that doesn't have the values, their code will crash.

It also doesn't allow you to modify paths without recompiling. You will still have to deploy code after changes to these files. So your workflow would just change from code edit -> deploy to pathplanner edit -> deploy. Unless c++ code deploy uses a cached build if code doesn't change, which I'm not sure about.

WarrenReynolds, 2022-03-24:
`but then if they change it to load a different path that doesn't have the values, their code will crash`. Sorry, I thought this stuff was part of the JSON file always, my bad.  Having said that, instead of the crashing the code with an exception, you could just return null or something similar if a JSON file was supplied that didn't have internal values for Vel and Acc and direction and the values weren't supplied in the constructor.  This then puts it back onto the programmers, if they supply paths without velocity and acceleration either internally in the JSON or in the call to loadPath in code the trajectory how can they expect them to work.  Which is basically the same as if you supply a JSON file with one point in the trajectory, this would work either.

`So your workflow would just change from code edit -> deploy to pathplanner edit -> deploy`

The reason I made this statement because I'm a bit scared from a recent competition we went to where last minute adjustments to autos couldn't be done because it just took to long to recompile and push. That's why I was so excited when 
I came across your implementation of the paths as text files in plan planner. I know java people never have to deal with these long build times, but if you check out Chief Delphi you will see an ever increasing number of people complaining about c++ build times.   To push a text file only takes 15 second, to recompile code and push it took us 15 minutes.

We have now addressed a heap of issues in our code that were causing excessive build times and now have it down to just under 5 minutes, but this is still massive compared to ftping a text file in 15 seconds.  I suppose I just wanted my cake and also wanted to eat it.  Anyway, its only a suggestion and your product is still amazing even if this request isn't implemented.

mjansen4857, 2022-03-24:
Yeah that makes more sense if you’re pushing the files over manually. 

If it were to just return a null value that would just push the exception further down the line and people would end up getting null pointer exceptions in their path followers. 

One suggestion I do have for avoiding having to deploy code just to change path vel/acc would be the [Preferences class](https://docs.wpilib.org/en/stable/docs/software/basic-programming/robot-preferences.html) This will let you edit path info through shuffleboard and then just restart code to generate everything with the updated values (assuming you’re loading paths when code starts instead of on the fly)

WarrenReynolds, 2022-03-24:
Thanks, haven't used preference before, but there has been heaps of times where we could have used this. So learning how do use preferences has just been push back to the top of the todo list.

viggy96, 2022-03-24:
> I'm a bit hesitant on this at the moment. These values are not guaranteed to be in path files and should not really be considered part of a path's definition like waypoints are. These values are [...]

Then an update to the PathPlanner GUI seems to be required here, that forces users to put in velocity/acceleration values for all paths. That way you're guaranteed to have those values available to pathplannerlib when the file is being read.

mjansen4857, 2022-07-02:
Ok so the way I've decided to do this is add a "getConstraintsFromPath" method that will load the max velocity and acceleration from the path file. I really want to make sure someone doesn't accidentally try to load a path without providing constraints unless they are sure that they want to do that. I think it would be fairly easy for someone without a lot of programming experience to call a load path method without constraint arguments (this would probably be the first code completion suggestion that VSCode provides) and not understand why it isn't working.

So, now this can be done but it will look more like this: 
`PathPlanner.loadPath("path name", PathPlanner.getConstraintsFromPath("path name"))`

It's a bit more verbose than just `loadPath("path name")` but you can always make your own helper method to make it simpler:
```java
public static PathPlannerTrajectory loadPath(String name){
    return PathPlanner.loadPath(name, PathPlanner.getConstraintsFromPath(name));
}
```

closed 2022-07-02 by mjansen4857

viggy96, 2022-10-13:
Has this been released to the Maven repo?

mjansen4857, 2022-10-13:
Not yet. Waiting for a WPILib beta before doing a PPLib beta release.