mjansen4857/pathplanner #73: Can "Path Group" function be added

mendax1234, 2022-03-19:
l think during an autonomous period, different paths is need, so a path group is needed. Since l want to do different actions during each path, so my method is to divide a whole path into several trajectories and decide which action the robot should do in each trajectory. But if doing so, l will have many trajectories, l think a path group will solve this inefficiency. Or , is there any other ways to do this?

JosiahFu, 2022-04-07:
Agreed, this would make autos easier to plan and sequence. My team's solution was to duplicate a master path several times and delete the extra points to create several sub-paths, then run the sub-paths and subsystem actions in sequence. However, separating the paths was tedious and could have caused problems when the endpoints did not match up.

Perhaps we could mark waypoints as "dividing points", and the path would be split into sub-paths between each dividing point? No new files would be created, but PathPlannerLib could reference each sub-path within the file individually. If this is too complex to implement, maybe it would just create new files for the sub-paths instead. That would still reduce human error.

closed 2022-05-20 by mjansen4857

mendax1234, 2022-05-20:
@mjansen4857 Thanks for adding this function, l am curious when the latest version of PathPlanner will be released?