mjansen4857/pathplanner #77: PathPlanner crashes upon opening path

5431-programming, 2022-03-21:
Using 2022.1.1, the app crashes when I open the app, click the menu, and click on another existing path.

mjansen4857, 2022-03-21:
Can you send the contents of the file that causes the crash?

jlmcmchl, 2022-03-23:
I have a similar crash on Windows 11, except in this case 2021.1.1 crashes when opening the side drawer. I can modify the existing path and simulate it, though. Strangely, the deploy button is missing entirely:
 [image]

mjansen4857, 2022-03-23:
I'm assuming that you're using the Microsoft Store version which is why the deploy button is missing as its supposed to be disabled for store builds. In the apple App Store it is impossible to get the permissions necessary to use the deploy button and I assumed Microsoft was the same but was not entirely sure since their documentation on this stuff isn't great. It was enabled in the last version on accident but if it was working for you I can re-enable it for Microsoft store builds.

As for the crash, I can't reproduce it so the only way for me to look into it is if you send me the stack trace for the crash. Unfortunately, flutter currently does not have a good way to log stuff like this so you'll have to install flutter, clone the repo and run the app in debug mode so you can get a stack trace for the crash.

mjansen4857, 2022-03-23:
They're the exact same thing except for one boolean flipped in home_page.dart line 54. Flipping that boolean shouldn't change anything that could lead to that unless something really weird is happening. If you uninstall the app from the windows store and then reinstall, does the crash persist?

jlmcmchl, 2022-03-23:
Repair, Reset, and Uninstall/Reinstall all left me with the same behavior. The crash wasn't always at the same time - sometimes the drawer would open, but sometimes it would crash upon clicking the button. Is there maybe a config file or something on my laptop that is causing this issue?

mjansen4857, 2022-03-23:
Can you try removing all of your paths from the project? Back them up in a folder on desktop or something so you have them. Then open the app and see if it works with the empty project, try adding new paths etc.

jlmcmchl, 2022-03-23:
looks like I was able to replicate it by adding two paths from our project in - here's the steps I took:

deleted all paths from project, open pathplanner.
- status: works fine, shows default 3-point path.
closed pathplanner. undeleted `01 - Line.path`, and reopened pathplanner
- status: opens fine, drawer is fine. only shows this path in drawer.
closed pathplanner. undeleted `02 - Two Ball.path`, and reopened.
- status: opens fine, shows `01 - Line.path`. Crashes upon opening the drawer.

Here's a gist with both paths if you can recreate it: https://gist.github.com/jlmcmchl/e29e73ce45841d31979ccf321a0a398b

mjansen4857, 2022-03-23:
Even with those paths I’m unable to replicate the crash. Weird. Only other thing I can think of would be to open a different project and then try switching back.

jlmcmchl, 2022-03-23:
That still causes a crash for me. I'm not worried about it - bummed I can't use the windows store to get updates but being able to use the github release is perfectly acceptable. Thanks for your help trying workarounds!

jlmcmchl, 2022-03-23:
It gets weirder - I think this is a OneDrive issue. When pathplanner is installed under %User%/OneDrive/Desktop/pathplanner-windows, I can replicate the crash - but under C:\Code\pathplanner-windows it works fine.

mjansen4857, 2022-03-23:
Interesting. I can't replicate this either. If you clone the repo into your onedrive folder and run it in debug mode can you reproduce it then?

jlmcmchl, 2022-03-23:
I was able to replicate it alright, but I can't get a consistent reason why it's crashing. This time, the first thing I did was open up the drawer, and this was the crash:
```[ +479 ms] flutter: Current Version: 2022.1.1, Latest Release: 2022.1.1
[+1737 ms] [ERROR:accessibility_bridge.cc(76)] Failed to update ui::AXTree, error: 26 will not be in the tree and is not the new root
[   +1 ms] [ERROR:accessibility_bridge.cc(76)] Failed to update ui::AXTree, error: 26 will not be in the tree and is not the new root
[  +27 ms] [ERROR:accessibility_bridge.cc(76)] Failed to update ui::AXTree, error: 26 will not be in the tree and is not the new root
[   +1 ms] [ERROR:accessibility_bridge.cc(76)] Failed to update ui::AXTree, error: 26 will not be in the tree and is not the new root
[+52114 ms] Service protocol connection closed.
[   +2 ms] Lost connection to device.
```

[image]

jlmcmchl, 2022-03-23:
While I can't demonstrate that all these crashes are the same, this seems to be tracked upstream, https://github.com/flutter/flutter/issues/98099. I don't think there's anything we can do here aside from maybe downgrade flutter? I haven't done anything to test that outcome, but can if that would help.

mjansen4857, 2022-03-24:
Yeah that definitely seems like the issue. That’s really unfortunate since 2.10 is a really great update for windows. I’ll replace the windows builds with 2.8 when I get a chance.

mjansen4857, 2022-03-24:
I have replaced the build in the GitHub release and the windows store release is in the submission process. Can you verify that this new build prevents the crashes?

jlmcmchl, 2022-03-25:
Downloaded a copy from the release page, works like a charm. Thanks!

mjansen4857, 2022-03-26:
Ok the store version should be updated as well. I'll leave this issue open for a while so anyone else with this problem can see they just need to update again.

mjansen4857, 2022-05-14:
@jlmcmchl Here is a build made with flutter 3.0 and using a normal list view in the drawer instead of a reorderable one:
[pathplanner-flutter3.zip](https://github.com/mjansen4857/pathplanner/files/8692008/pathplanner-flutter3.zip)

If you get a chance, can you see if this crash has been resolved with these changes?

mjansen4857, 2022-07-03:
Everything I can find on the issue points to it being a reorderable list view problem, which is no longer used so this shouldn't be an issue anymore. Feel free to re-open if it is.

closed 2022-07-03 by mjansen4857