mjansen4857/pathplanner #78: Expected a double but reading an int in the .path files crashes program (NB:Not a fault of pathfinder)

WarrenReynolds, 2022-03-27:
I found a workflow that crashes pathfinder.  Not really the programs fault if someone manually changes the data inside the .path JSON file by using another external program, but I think this potential problem can be fixed with a mirror change to the source and it will make pathfinder more resilient to malformed .path files.

I found this issue when I tried a JSON viewer/editor that showed JSON files in a friendly tree view instead of one continuous line.
If you view the .path JSON files created by pathfinder in VScode and use a JSON viewer/editor extension like JSON Tools and then you make a change to the JSON file and then save the JSON.  The change will be reflected back in pathplanner when you open it again. (This is great for making small adjustments, but!)

The issue I had was I had a holomonic angle of 0.0 in one of my paths.  When I used the JSON Tool extension (from VScode) to format the view into a friendly tree view it changed the 0.0 to just 0  
Then when I saved this file and tried to open pathfinder again it crashed the program.  On doing a bit of research I believe the issue is the JSON parser inside pathplanner is seeing the 0 as an int and it is expecting to read a double and this prevents pathplanner from loading.

Now, I know this issue will never present itself if you only ever make changes to the paths inside pathfinder as it will always create and read doubles even if the user enter the values as integers.  But if the paths are saved in a text format like JSON, people are going to look/ edit these files in text editors. So a simple change to one line of code could prevent this issue if a person was to manually edit the .path json file and accidental leave off the .0 on a number that was supposed to be a double.

I found this also happens on the max vel, and max acc values, and I assume the manual vel override on each waypoint (but I didn't test this). 

If Line 294 of waypoint.dart was changed from
holonomicAngle = json ['holonomicAngle'];
to holonomicAngle = double.parse(json['holonomicAngle'].toString());

I think this would fix this possible situation where an external change to a .path file could crash the program.
Basically this code change would need to be applied where ever a double is expect but someone could accidentally enter an int.

Again this is not really a bug in the pathfinder program and you have every right to say, not my problem if someone modifies these files outside of pathfinder and they don't enter the data properly. But I think the fact that you can easily change the path if needed by editing text is an excellent feature, and this change will only make the program a better program.

mjansen4857, 2022-03-27:
Will definitely fix this, its a pretty simple fix. Dart supports the type `num` which basically acts like both `int` and `double`. Changing the types of all values currently stored as doubles to num should fix this and have the side affect of `0.0` just being stored as `0` as well.

mjansen4857, 2022-03-27:
Fixed in [e93d252](https://github.com/mjansen4857/pathplanner/commit/e93d2524059a9c02f20eb4016bc2dd0b24b1e08a)

I guess it still stores 0 as 0.0 but I did test manually changing values to integers and the app launched fine so this does fix it. I don't think this is a big enough issue to push a whole update for, but it will be included whenever the next one is released. You could alway build manually if you want the fix now. 

Thanks for pointing this out!

closed 2022-03-27 by mjansen4857

WarrenReynolds, 2022-03-27:
I'll give it a crack at building manually from your source.  I've never used flutter before so I'm not holding much hope. But you are never to old to learn so why not try.  Thanks again for sharing this wonderful program with the FRC community.

WarrenReynolds, 2022-03-27:
I just built from your source and found another issue while testing.  If a waypoint coordinate in the path is changed to an integer, like in my test path that goes from (0.41,3) to (1.41,3) the program will load ok (NB: In my case the path that loads when pathplnner loads is not the path with integers co-ords in it) So this indicates that first bug is fixed ie the program now loads. But if you change to the path (that has integer coordinates) and then try to click on one of the waypoints to view the angles etc, the screen turns white and the program hangs.  Maybe the co-ordinates also need to be changed to the num datatype to handle this case.

mjansen4857, 2022-03-27:
The coordinates should already be stored as the num data type. If you run the program in debug mode: `flutter run` it should give you a stack trace.

WarrenReynolds, 2022-03-27:
Does this help?
[image]

`When the exception was thrown, this was the stack:
#0      Waypoint.getYPos (package:pathplanner/robot_path/waypoint.dart:65:26)
#1      _WaypointCardState._buildPositionRow (package:pathplanner/widgets/path_editor/waypoint_card.dart:148:43)
#2      _WaypointCardState.build (package:pathplanner/widgets/path_editor/waypoint_card.dart:54:21)
#3      StatefulElement.build (package:flutter/src/widgets/framework.dart:4870:27)
#4      ComponentElement.performRebuild (package:flutter/src/widgets/framework.dart:4754:15)
#5      StatefulElement.performRebuild (package:flutter/src/widgets/framework.dart:4928:11)
#6      Element.rebuild (package:flutter/src/widgets/framework.dart:4477:5)
#7      StatefulElement.update (package:flutter/src/widgets/framework.dart:4960:5)
#8      Element.updateChild (package:flutter/src/widgets/framework.dart:3501:15)
#9      SingleChildRenderObjectElement.update (package:flutter/src/widgets/framework.dart:6291:14)
#10     Element.updateChild (package:flutter/src/widgets/framework.dart:3501:15)
#11     RenderObjectElement.updateChildren (package:flutter/src/widgets/framework.dart:5787:32)
#12     MultiChildRenderObjectElement.update (package:flutter/src/widgets/framework.dart:6445:17)
#13     Element.updateChild (package:flutter/src/widgets/framework.dart:3501:15)
#14     ComponentElement.performRebuild (package:flutter/src/widgets/framework.dart:4780:16)
#15     Element.rebuild (package:flutter/src/widgets/framework.dart:4477:5)
#16     ProxyElement.update (package:flutter/src/widgets/framework.dart:5108:5)
#17     Element.updateChild (package:flutter/src/widgets/framework.dart:3501:15)
#18     SingleChildRenderObjectElement.update (package:flutter/src/widgets/framework.dart:6291:14)
#19     Element.updateChild (package:flutter/src/widgets/framework.dart:3501:15)
#20     ComponentElement.performRebuild (package:flutter/src/widgets/framework.dart:4780:16)
#21     StatefulElement.performRebuild (package:flutter/src/widgets/framework.dart:4928:11)
#22     Element.rebuild (package:flutter/src/widgets/framework.dart:4477:5)
#23     StatefulElement.update (package:flutter/src/widgets/framework.dart:4960:5)
#24     Element.updateChild (package:flutter/src/widgets/framework.dart:3501:15)
#25     ComponentElement.performRebuild (package:flutter/src/widgets/framework.dart:4780:16)
#26     Element.rebuild (package:flutter/src/widgets/framework.dart:4477:5)
#27     ProxyElement.update (package:flutter/src/widgets/framework.dart:5108:5)
#28     Element.updateChild (package:flutter/src/widgets/framework.dart:3501:15)
#29     SingleChildRenderObjectElement.update (package:flutter/src/widgets/framework.dart:6291:14)
#30     Element.updateChild (package:flutter/src/widgets/framework.dart:3501:15)
#31     ComponentElement.performRebuild (package:flutter/src/widgets/framework.dart:4780:16)
#32     StatefulElement.performRebuild (package:flutter/src/widgets/framework.dart:4928:11)
#33     Element.rebuild (package:flutter/src/widgets/framework.dart:4477:5)
#34     StatefulElement.update (package:flutter/src/widgets/framework.dart:4960:5)
#35     Element.updateChild (package:flutter/src/widgets/framework.dart:3501:15)
#36     ComponentElement.performRebuild (package:flutter/src/widgets/framework.dart:4780:16)
#37     StatefulElement.performRebuild (package:flutter/src/widgets/framework.dart:4928:11)
#38     Element.rebuild (package:flutter/src/widgets/framework.dart:4477:5)
#39     BuildOwner.buildScope (package:flutter/src/widgets/framework.dart:2659:19)
#40     WidgetsBinding.drawFrame (package:flutter/src/widgets/binding.dart:882:21)
#41     RendererBinding._handlePersistentFrameCallback (package:flutter/src/rendering/binding.dart:363:5)
#42     SchedulerBinding._invokeFrameCallback (package:flutter/src/scheduler/binding.dart:1144:15)
#43     SchedulerBinding.handleDrawFrame (package:flutter/src/scheduler/binding.dart:1081:9)
#44     SchedulerBinding._handleDrawFrame (package:flutter/src/scheduler/binding.dart:995:5)
#48     _invoke (dart:ui/hooks.dart:151:10)
#49     PlatformDispatcher._drawFrame (dart:ui/platform_dispatcher.dart:308:5)
#50     _drawFrame (dart:ui/hooks.dart:115:31)
(elided 3 frames from dart:async)`

WarrenReynolds, 2022-03-28:
Sorry missed this part at the top of the stack trace
`══╡ EXCEPTION CAUGHT BY WIDGETS LIBRARY ╞═══════════════════════════════════════════════════════════
The following _CastError was thrown building WaypointCard(dirty, state: _WaypointCardState#36c42):
type 'int' is not a subtype of type 'double' in type cast

The relevant error-causing widget was:
  WaypointCard
  WaypointCard:file:///C:/Users/wreynolds5.DETNSW/Downloads/Compressed/pathplanner-main/pathplanner-main/lib/widgets/pat
  h_editor/path_editor.dart:408:14`

mjansen4857, 2022-03-28:
I think I fixed it in the latest commit

WarrenReynolds, 2022-03-28:
Cheers,  I'll build again and test.  I'm not much of a programmer, but at least I can try to contribute by trying to make the code crash.

WarrenReynolds, 2022-03-28:
I deleted the pathplanner folder, re downloaded from github, an then rebuilt the exe.
It still seems to break when changing to the path with integer coordinates.
I tried to debug with  'flutter run' and this now happens when I attempt to change to the problem path, so I can't even give you a stack trace.
[image]
If I press retry the window closes and this is whats left in the cmd window.
[image]

mjansen4857, 2022-03-28:
This is a different bug in flutter 2.10. Downgrading your flutter version to 2.8.1 will fix this. I’ve also seen some say that installing the latest c++ redistributable from [here](https://docs.microsoft.com/en-us/cpp/windows/latest-supported-vc-redist?view=msvc-170) can fix it but I haven’t been able to confirm that since that bug doesn’t happen on my system.