mjansen4857/pathplanner #81: Add python install to wiki

virtuald, 2022-04-03:
Wiki doesn't seem to be editable? Here's the link: https://robotpy.readthedocs.io/en/stable/install/pathplannerlib.html

Thanks!

mjansen4857, 2022-04-03:
Added the link. Thanks for putting this together. Is there any usage documentation yet? I haven't found any. I can link to that as well if you have some.

mjansen4857, 2022-04-03:
Nevermind I just found it

closed 2022-04-03 by mjansen4857

virtuald, 2022-04-03:
@TheTripleV put together the initial wrapper around this, not sure if they were planning on translating the wiki for the python documentation page or not. But, it's pretty close to Java usage.

mjansen4857, 2022-04-30:
Just as an FYI I moved the pathplannerlib code into the main repo instead of a separate repository. I'm not sure if this affects the python library but letting you know just in case.

virtuald, 2022-04-30:
I just use whatever is published in the sources.zip, so it doesn't matter to me.