mjansen4857/pathplanner #82: Problems when sign of Holonomic Angle Changes

kinahawi, 2022-04-06:
Thanks for providing this capability! We're having pretty good success using the holonomic mode with our mecanum drive bot but one thing is tripping us up. When the holonomic angle changes sign from one waypoint to the next, it causes the bot to lose path following. For example, if the first way point has an angle of 79 degrees and the next one has an angle of -145, the bot goes out of control. When I examine the states generated from the waypoints, I see the angle grow from 79 degrees up to about 215, then it jumps to -145 and proceeds from there according to the control points and next way point. The math makes sense from the perspective that its driven by 360 degrees in a circle, but I'm concerned that the PID for theta is getting messed up when it's calculate method is given a reference angle that's significantly negative and a current pose angle that is significantly positive. I don't think the profiled PID is cognizant that the values are angles and probably just does simple math on them, not accounting for the circular nature of the numbers.

Perhaps there is something else going on, or you have a suggestion for something I may be doing wrong? Any help is greatly appreciated as we're preparing for our second event this weekend.

bruingineer, 2022-04-06:
Make sure the PID controller is in continuous mode

mjansen4857, 2022-04-06:
Can you send the exact path file that causes this? The holonomic angle should always be between -180 and 180 so if it’s generating numbers outside of that then it’s not intended.

kinahawi, 2022-04-06:
Here is a path that produces angles that are outside -180..180. I've also included a .csv file which shows the angles from the states after the path is loaded
[Data for Path Planner Issue 86.zip](https://github.com/mjansen4857/pathplanner/files/8426426/Data.for.Path.Planner.Issue.86.zip)
.

mjansen4857, 2022-04-07:
I'll take a look. In the meantime, it sounds like enabling continuous input will still fix this. It expects radians so make sure you do it between -pi and pi.

mjansen4857, 2022-05-11:
Fixed this in [074bd70](https://github.com/mjansen4857/pathplanner/commit/074bd7062403d169515692eb96c9064643bcbb51)