mjansen4857/pathplanner #83: Error LNK2019: unresolved external symbols when building robot on windows

THRYLLR, 2022-04-08:
In our robot's code, we have a function:
```cpp
frc::Trajectory Drivetrain::GetAutoTrajectory() {
  PathPlannerTrajectory autonomousPath = PathPlanner::loadPath("New Path", 4_mps, 4_mps_sq);
  frc::Trajectory trajectory = autonomousPath.asWPILibTrajectory();
  return trajectory;
}
```
which retrieves a path file from our robot's deploy directory and returns it in `frc::Trajectory` form. I am not entirely sure the below error is related to errors in our code, but upon building our robot's code we get two fatal linker errors saying that `asWPILibTrajectory` and `loadPath` are unresolved external symbols.

We have attempted reinstalling WPILib, PathPlanner and pathplannerlib; and we have checked for typical errors such as type mismatches or misspellings between headers and sources. We are out of ideas and we have a regional event which starts tomorrow.

Might this error be one inherent to my machine (Windows 10, WPILib 2022, WPILib VSCode)?

```
> Task :linkFrcUserProgramWindowsx86-64ReleaseExecutable FAILED
Drivetrain.obj : error LNK2019: unresolved external symbol "public: class frc::Trajectory __cdecl pathplanner::PathPlannerTrajectory::asWPILibTrajectory(void)" (?asWPILibTrajectory@PathPlannerTrajectory@pathplanner@@QEAA?AVTrajectory@frc@@XZ) referenced in function "public: class frc::Trajectory __cdecl Drivetrain::GetAutoTrajectory(void)" (?GetAutoTrajectory@Drivetrain@@QEAA?AVTrajectory@frc@@XZ)
Drivetrain.obj : error LNK2019: unresolved external symbol "public: static class pathplanner::PathPlannerTrajectory __cdecl pathplanner::PathPlanner::loadPath(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,class units::unit_t<struct units::unit<struct std::ratio<1,1>,struct units::base_unit<struct std::ratio<1,1>,struct std::ratio<0,1>,struct std::ratio<-1,1>,struct std::ratio<0,1>,struct std::ratio<0,1>,struct std::ratio<0,1>,struct std::ratio<0,1>,struct std::ratio<0,1>,struct std::ratio<0,1> >,struct std::ratio<0,1>,struct std::ratio<0,1> >,double,struct units::linear_scale>,class units::unit_t<struct units::unit<struct std::ratio<1,1>,struct units::base_unit<struct std::ratio<1,1>,struct std::ratio<0,1>,struct std::ratio<-2,1>,struct std::ratio<0,1>,struct std::ratio<0,1>,struct std::ratio<0,1>,struct std::ratio<0,1>,struct std::ratio<0,1>,struct std::ratio<0,1> >,struct std::ratio<0,1>,struct std::ratio<0,1> >,double,struct units::linear_scale>,bool)" (?loadPath@PathPlanner@pathplanner@@SA?AVPathPlannerTrajectory@2@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V?$unit_t@U?$unit@U?$ratio@$00$00@std@@U?$base_unit@U?$ratio@$00$00@std@@U?$ratio@$0A@$00@2@U?$ratio@$0?0$00@2@U32@U32@U32@U32@U32@U32@@units@@U?$ratio@$0A@$00@2@U52@@units@@NUlinear_scale@2@@units@@V?$unit_t@U?$unit@U?$ratio@$00$00@std@@U?$base_unit@U?$ratio@$00$00@std@@U?$ratio@$0A@$00@2@U?$ratio@$0?1$00@2@U32@U32@U32@U32@U32@U32@@units@@U?$ratio@$0A@$00@2@U52@@units@@NUlinear_scale@2@@7@_N@Z) referenced in function "public: class frc::Trajectory __cdecl Drivetrain::GetAutoTrajectory(void)" (?GetAutoTrajectory@Drivetrain@@QEAA?AVTrajectory@frc@@XZ)
C:\.......\Robot2022\build\exe\frcUserProgram\windowsx86-64\release\frcUserProgram.exe : fatal error LNK1120: 2 unresolved externals
```

mjansen4857, 2022-04-08:
I have no idea. If you create a brand new project and install pathplannerlib in that project will it build properly with this function? That could narrow it down to being an issue with your project configuration or something.

THRYLLR, 2022-04-08:
I just created a project and had it compile the same function that threw linker errors in my original project. Assuming I wrote the test project correctly, it compiled successfully without any errors. I'm not sure if it's related to the configuration of our original project, but I'll do some careful inspection of it.

THRYLLR, 2022-04-08:
Fortunately, I successfully resolved this error just in time for our team's competition. The problem was that this library apparently doesn't compile correctly on Windows. However, our Gradle settings were set to attempt to compile the Windows version of our code for simulation purposes. We did not need to simulate anything, so we disabled the setting and now we're good to go!

Thanks for the help

closed 2022-04-08 by THRYLLR

mjansen4857, 2022-04-08:
Yeah that makes sense. The library doesn’t support windows simulation and I can’t get it to compile for windows for some reason. Hopefully I can figure that out by next season. Glad you got it worked out.