mjansen4857/pathplanner #84: Trajectory Initial Starting Velocity

J-Barta, 2022-04-08:
I can't find a way to make my trajectories generate with an initial starting velocity, which is very important if I wanted to start generating a trajectory on-the-fly with a potential starting linear and/or angular velocity. How would I go about implementing something like this?

mjansen4857, 2022-04-08:
I think you’re asking about pathplannerlib specifically. The GUI has a velocity override field on each waypoint that can do this, but pathplannerlib currently doesn’t really support any on the fly generation. This is definitely something I want to support in a future release.

mjansen4857, 2022-09-12:
I think this was taken care of in #114 

I could add overrides to the loadPath methods but I don't think they'd ever really be used since if you're doing on the fly generation you're likely going to generate a path from your current odometry position to something like a game piece which you can't pre-make in the GUI

closed 2022-09-12 by mjansen4857