mjansen4857/pathplanner #93: Moving waypoints close to each other results in rapid spins

woofer300, 2022-07-04:
Hello! I am having an issue with the PathPlanner app. When I move two waypoints in a path too close to each other, it results in the trajectory having rapid 180 degree spins. Is this an intentional design? How can I make it so that the robot follows the path without spinning? Thanks!

https://user-images.githubusercontent.com/66227348/177220151-eb1e383a-f78c-4a25-85a8-692e18d0a810.mp4

mjansen4857, 2022-07-04:
It's not "intentional" per say but it is how bezier curves work. Those little dots connected to the anchor points are called control points and they sort of "pull" the path towards them. In the situation you recorded, you have the control point behind the other anchor point, so the path it ends up creating doubles back on itself a little bit. Here's a quick drawing of what I mean in a non-straight line so you can see what I'm talking about: 
[image]

The way to fix this is to make the control points shorter:
[image]

There's a lot of quirks you'll run into with this but they can pretty much all be solved by tweaking the control points a bit.

woofer300, 2022-07-04:
Wow I'm stupid, thanks so much for the help.

closed 2022-07-04 by woofer300