mjansen4857/pathplanner #107: AUR package, pathplanner.desktop file

isaec, 2022-09-02:
I created an [aur package](https://aur.archlinux.org/packages/pathplanner-bin) for pathplanner. It would be nice if the ``.desktop`` file that my PKGBUILD generates was included somehow in the repository.
```
[Desktop Entry]
Version=1.5
Type=Application
Name=PathPlanner
Exec=/usr/bin/pathplanner
Categories=Development
Keywords=FRC;Motion Profile;Path Planning
Icon=/opt/pathplanner-bin/data/flutter_assets/images/icon.png
```
This could definitely be improved, just what I threw together.

It could also be good to list the aur package as a place to install and get updates from - if you see issues with the package, please let me know as I am quite new to aur packaging.

Thanks for making such a useful tool!

mjansen4857, 2022-09-12:
I'm not sure about adding a file for a package I don't maintain to the main repo. Perhaps adding it to your fork would be better?

I'll add a link to the package in the readme.

mjansen4857, 2022-09-12:
Although I might be misunderstanding aur as I don't use arch. Is it a package manager like apt etc. or just the sources for building manually? I guess I mainly don't understand what including this file in the repo would do for it.

isaec, 2022-09-13:
I'm not saying you should include the whole build file -- I was just saying you should include the `.desktop` file somewhere.

Here is the spec, although you certainly don't need to read it. <https://specifications.freedesktop.org/desktop-entry-spec/desktop-entry-spec-latest.html>

It's not a big deal, but using the same `.desktop` file everywhere could allow greater consistency for linux users?

The PKGBUILD is a build script for an existing file. The aur is a collection of user submitted build scripts for various files. Usage can be automated, with a tool like `yay` or `paru`, among others.

For arch users, installing a PKGBUILD makes it easy to keep track of files and do updates, like apt.

One of the steps in my PKGBUILD is to create a `.desktop` file that I threw together, so that pathplanner can be launched graphically.

I actually can't find many examples of a repo including a `.desktop` file, so unless you want to add one to the snap package, it's probably fine as is -- sorry to bug you.

mjansen4857, 2022-09-14:
No problem, thanks for the explanation. I don't update the snap package anymore since the number of people using it was fairly small to justify the extra effort of maintaining it. So I suppose it is probably best to leave as is then.

isaec, 2022-09-14:
Keeping a binary PKGBUILD up to date is usually super easy so I should keep it up to date for the time being. Thanks for the tool!

closed 2022-09-14 by isaec