mjansen4857/pathplanner #108: Add the ability to plot the velocity/acceleration/etc. for the generated trajectory

bckircher, 2022-09-04:
In a new editor, plot the appropriate set of traces (velocity, acceleration, heading, angular velocity, curvature for non-holonomic robots, velocity, acceleration, rotation, and angular velocity for holonomic robots) so that the commanded behavior of the robot can be viewed. The editor would look something like the following:

[image]

Traces could be turned on or off by clicking on them in the legend at the bottom, allow certain traces to be isolated (or uncovered if traces overlapped at some point of interest).

bckircher, 2022-09-05:
The initial version of this is at https://github.com/bckircher/pathplanner/tree/graph_editor. @mjansen4857  when you have a chance, try it out and let me know what you think. Both UX and code structure. Once we're both happy with it, I'll create a PR.

Note there is one somewhat unrelated change in here right now, which I'm happy to pull out separately if that is better. The click radius for markers was reduced from 40 to 15 (which is about the size of the marker itself). At 40, it was really hard to select markers that were located close to each other (it was easy to "hide" a marker behind the large click radius of another marker). It's set to 15 in the new graph editor, and I updated it to 15 in the marker editor to be consistent.

The graph editor code feels a bit icky to me, but I'm not sure if that is the code itself, my still relative unfamiliarity with Dart/Flutter, or a combination of both!

mjansen4857, 2022-09-12:
Haven't gotten to play with it yet, but I just merged a PR which updates flutter to 3.3.1 and changes the linter configuration to find a lot more best practice stuff. I had to fix 300+ issues so there will probably be some in this branch if you copied anything. All you should have to do is make sure you're updated to 3.3.1 and then go through the "Problems" tab in VSCode and the quick fix option it suggests should be good. If there's anything you're unsure of let me know.

I will probably make changes to this when I get around to #97 to show target vs actual values when running path following commands. If I see anything that can be improved I'll probably just do it then. There's likely a package on https://pub.dev/ that can be used to make a graph without having to paint everything manually so between now and then I'll look around for one that can match the features this has since using a full featured graph widget will be a lot easier to maintain.

bckircher, 2022-09-12:
OK, makes sense. I'll pull in the latest changes to this branch (and take care of any lint errors I'm responsible for!), then see what I can find in the flutter package repo to redo the graph.

mjansen4857, 2022-09-12:
I've used this in a different project https://pub.dev/packages/fl_chart

You don't have to worry about it now if you don't want to. This is a good first iteration that will work well until it needs to be expanded.

bckircher, 2022-09-20:
FL Chart looks nice. I'm looking into using it instead of the stuff I did...

closed 2022-10-20 by mjansen4857