mjansen4857/pathplanner #111: Updating a path while Excel has CSV file open crashes Path Planner

WarrenReynolds, 2022-09-05:
I can live with this problem, as I have learned what causes it, but crashing path planner without warning is something that probably should be fixed if it can be.

If you are looking at one of the CSV files generated by pathplanner in Excel. Then you make an adjustment to a path in pathplanner, this will crash pathplanner without warning.  I assume this is because Excel has an open handle on the file and pathplanner can't open a writable handle to the CSV file.  Maybe displaying a warning message like "Unable to update path because another program has a file open in the deploy folder. Your change to the path has been undone. Please close the external program to enable changes to paths" instead of crashing the whole program.

mjansen4857, 2022-09-05:
Are you using the release version or the current main branch? There have been changes since the last release that should stop it from crashing, but I suppose showing the user that writing a file failed is a good idea.

WarrenReynolds, 2022-09-05:
The last build I did was back on the 14 Aug, I'll download the latest from github and test again.

mjansen4857, 2022-09-05:
That should have had those changes then. Not sure why it crashed but I'll look into it

WarrenReynolds, 2022-09-06:
I just re downloaded and built.  You are correct, the current version doesn't crash the program if excel has one of the csv files open.  The only issue I can see now is that the CSV just doesn't get updated if its open in another program, but there is no warning that states this.  So I think adding the warning message would be good so that you know you have to close the excel file is you want the latest changes to flow into the CSV file.

closed 2022-09-12 by mjansen4857

WarrenReynolds, 2022-09-20:
Hi Michael,
I'm not sure if this is where I should say this or if I should open a new Issue, but I just downloaded and built the latest version of pathplanner and I don't think this issue has been resolved.  To me, it still looks like the csv file that excel has open doesn't get updated when a path is adjusted and there isn't any warning displayed in the GUI that tells you this.  I wish I new a lot more about programming flutter and dart so I could investigate the issue further but unfortunately all I can do is point out the issues I find. 
Regards
Warren

mjansen4857, 2022-09-21:
I’ll play around with it again but I’m not sure why it wouldn’t be showing an error. It should now show one on any exception while writing the file.