mjansen4857/pathplanner #113: Tab key doesn't save values before moving the focus.

WarrenReynolds, 2022-09-06:
Not really an issue or bug, more of a feature request. 
I always use the TAB change to move to the next field after updating a value.  At the moment if you do this the value in a text box doesn't get updated unless you press the ENTER key.  I'm so used to the ENTER key being used to accept the current change and close the dialog that I never press it when updating values, so I find myself always making mistakes in this area.

I don't know what Flutter's default behaviour is around dialog boxes, but if an options exists to make the TAB key update the value before it moves the focus to the next field this would be a nice option to have in the settings for path planner.

This is only a suggestion to make a great program even better.

mjansen4857, 2022-10-17:
So I looked into this and there's currently a bug with flutter that prevents me from knowing that the tab key was pressed (in order to update the path value) while still maintaining the behavior of the tab key switching focus to the next text field.

I'll come back to this once that gets fixed.

isaec, 2022-10-30:
Alternatively, after a bit of debouncing changes could be written to file without waiting for the focus to switch?

Seems slightly counterintuitive to only write changes when a new element is focused.

mjansen4857, 2023-08-19:
This will now work for number text fields after #328. But, I did not do this for any name text fields since renaming files and such is something I want to have intent behind it to prevent you from renaming a file to something weird when you click off of a text field and not notice that the name changed.

closed 2023-08-19 by mjansen4857