mjansen4857/pathplanner #116: Unable to access paths on mac

Ashray-g, 2022-09-23:
The application opens blurry on mac (tried download from git multiple times and app store a couple times)

[image]

mjansen4857, 2022-09-23:
Is this the first time you’ve used the app or has it worked fine before?

Ashray-g, 2022-09-23:
It worked fine all year for us until about a week ago.

mjansen4857, 2022-09-23:
My guess then is that one of your path files is malformed and causing a crash. Try removing all of the paths from the project you had opened then see if it opens up fine. If so you can try slowly adding paths back until it fails to open again or just open them in notepad and see if there’s anything obvious.

Ashray-g, 2022-09-23:
Oh I might have also moved my main project folder to a different directory. Maybe it's having trouble since it's not where it was anymore?

Ashray-g, 2022-09-23:
Since I deleted + reinstalled the app why would it have the information on what project I'm currently using?

mjansen4857, 2022-09-23:
That would make sense. It will try to open the last folder that was opened so if that doesn’t exist anymore that could cause issues. If that ends up being the case I’ll switch this issue to making changes to allow the app to recover from failing to open a project. Uninstalling the app doesn’t actually remove the file that the settings are stored in. I don’t remember the exact path for the file. I’ll look for it.

Ashray-g, 2022-09-23:
Oh, awesome! Thanks for the insanely quick reply, we have chezy today actually and I just remembered to do this in case we need to make changes there.

Ashray-g, 2022-09-23:
Is it the plist file in Library/Preferences?

mjansen4857, 2022-09-23:
Run this in terminal: `defaults delete com.mjansen4857.pathplanner flutter.currentProjectDir`

Ashray-g, 2022-09-23:
Awesome! That worked

mjansen4857, 2022-09-23:
Great. I'll use this issue to fix it crashing then.

closed 2022-10-17 by mjansen4857