mjansen4857/pathplanner #117: Pathplanner doesn't launch on my computer

J-Barta, 2022-09-30:
Pathplanner will no longer launch on my Windows 11 computer. It definitely hasn't worked since updating to the newest Windows version (22H2), but I honestly don't know if the update caused it to not work. I know it was working some time (like a month) before updating, and it isn't working now (like a week after updating). I've tried uninstalling and re-installing from the windows store, as well as downloading the exe directly and running it, but the app didn't launch in either of those cases.

mjansen4857, 2022-10-01:
This is really hard to diagnose so can you install flutter and run the app in debug mode? If it still crashes this should give you some error output.

clmazzac, 2022-10-08:
i problemed

```Version=1
EventType=MoAppCrash
EventTime=133097318741353284
ReportType=2
Consent=1
UploadTime=133097318743476430
ReportStatus=268435456
ReportIdentifier=2facffe8-f903-421a-ae7a-bd3a00407ef6
IntegratorReportIdentifier=f289ba7b-cb2d-4d10-8c06-da4c59d5b13f
Wow64Host=34404
NsAppName=praid:pathplanner
OriginalFilename=pathplanner.exe
AppSessionGuid=00002b34-0002-0008-1058-ef634edbd801
TargetAppId=U:39956MichaelJansen.FRCPathPlanner_2022.1.2.0_x64__x7a2cjep50mda!pathplanner
TargetAppVer=2022.1.2.0_x64_!2022//03//24:18:53:48!0!pathplanner.exe
BootId=4294967295
TargetAsId=5602
IsFatal=1
EtwNonCollectReason=1
Response.BucketId=e3d86ba2568e1026875f9f1f8bc694fc
Response.BucketTable=5
Response.LegacyBucketId=1684239743497704700
Response.type=4
Sig[0].Name=Package Full Name
Sig[0].Value=39956MichaelJansen.FRCPathPlanner_2022.1.2.0_x64__x7a2cjep50mda
Sig[1].Name=Application Name
Sig[1].Value=praid:pathplanner
Sig[2].Name=Application Version
Sig[2].Value=1.0.0.0
Sig[3].Name=Application Timestamp
Sig[3].Value=623cbe3c
Sig[4].Name=Fault Module Name
Sig[4].Value=pathplanner.exe
Sig[5].Name=Fault Module Version
Sig[5].Value=1.0.0.0
Sig[6].Name=Fault Module Timestamp
Sig[6].Value=623cbe3c
Sig[7].Name=Exception Code
Sig[7].Value=c0000005
Sig[8].Name=Exception Offset
Sig[8].Value=0000000000005e50
DynamicSig[1].Name=OS Version
DynamicSig[1].Value=10.0.22621.2.0.0.768.101
DynamicSig[2].Name=Locale ID
DynamicSig[2].Value=1033
DynamicSig[22].Name=Additional Information 1
DynamicSig[22].Value=8eb0
DynamicSig[23].Name=Additional Information 2
DynamicSig[23].Value=8eb0c9bfca938091fe2837411a56a7fd
DynamicSig[24].Name=Additional Information 3
DynamicSig[24].Value=1ca1
DynamicSig[25].Name=Additional Information 4
DynamicSig[25].Value=1ca1d9edc73f4e5b27e3c2668d65ac00
UI[2]=C:\Program Files\WindowsApps\39956MichaelJansen.FRCPathPlanner_2022.1.2.0_x64__x7a2cjep50mda\pathplanner.exe
LoadedModule[0]=C:\Program Files\WindowsApps\39956MichaelJansen.FRCPathPlanner_2022.1.2.0_x64__x7a2cjep50mda\pathplanner.exe
LoadedModule[1]=C:\WINDOWS\SYSTEM32\ntdll.dll
LoadedModule[2]=C:\WINDOWS\System32\KERNEL32.DLL
LoadedModule[3]=C:\WINDOWS\System32\KERNELBASE.dll
LoadedModule[4]=C:\WINDOWS\System32\USER32.dll
LoadedModule[5]=C:\Program Files\WindowsApps\39956MichaelJansen.FRCPathPlanner_2022.1.2.0_x64__x7a2cjep50mda\file_selector_windows_plugin.dll
LoadedModule[6]=C:\Program Files\WindowsApps\39956MichaelJansen.FRCPathPlanner_2022.1.2.0_x64__x7a2cjep50mda\url_launcher_windows_plugin.dll
LoadedModule[7]=C:\WINDOWS\System32\win32u.dll
LoadedModule[8]=C:\WINDOWS\System32\GDI32.dll
LoadedModule[9]=C:\WINDOWS\System32\gdi32full.dll
LoadedModule[10]=C:\WINDOWS\System32\msvcp_win.dll
LoadedModule[11]=C:\WINDOWS\System32\ucrtbase.dll
LoadedModule[12]=C:\WINDOWS\System32\SHELL32.dll
LoadedModule[13]=C:\Program Files\WindowsApps\39956MichaelJansen.FRCPathPlanner_2022.1.2.0_x64__x7a2cjep50mda\flutter_windows.dll
[... 91 lines cut ...]
OsInfo[30].Value=8E409263-45B4-438D-AD01-6E9674AB032A.1
OsInfo[31].Key=edition
OsInfo[31].Value=Core
OsInfo[32].Key=ring
OsInfo[32].Value=Retail
OsInfo[33].Key=expid
OsInfo[33].Value=RS:1226D
OsInfo[34].Key=fconid
OsInfo[34].Value=18299130,0,2,0;19638787,0,2,0;23531064,2,2,0;23562335,2,2,1;23563673,2,2,0;23877894,1,2,0;25704915,1,2,0;26008830,0,1,1;29882130,0,1,1;29882131,0,1,1;34592303,0,2,1;35620393,0,2,1;35681102,0,1,0;36828252,1,2,0;37007953,0,1,1;37337255,0,2,1;37337272,0,2,1;37634385,0,2,1;37715785,0,2,1;38320543,0,1,1;38758896,0,2,1;38764045,0,1,1;38859191,0,1,1;38888144,0,1,1;39130862,0,2,1;39130896,0,2,1;39139876,0,2,1;39145991,0,1,1;39146010,0,1,1;39177456,0,2,1;39281392,0,1,1;39410888,0,2,1;40131846,0,1,0;40260922,0,1,0
OsInfo[35].Key=containerid
OsInfo[36].Key=containertype
OsInfo[37].Key=edu
OsInfo[37].Value=0
OsInfo[38].Key=servicinginprogress
OsInfo[38].Value=0
OsInfo[39].Key=featureupdatependingreboot
OsInfo[39].Value=0
FriendlyEventName=Stopped working
ConsentKey=MoAppCrash
AppName=A new Flutter project.
AppPath=C:\Program Files\WindowsApps\39956MichaelJansen.FRCPathPlanner_2022.1.2.0_x64__x7a2cjep50mda\pathplanner.exe
NsPartner=windows
NsGroup=windows8
ApplicationIdentity=577650238B5120865022B9F6B617F78B
MetadataHash=-1633586243```

ehamwey, 2022-10-12:
We (3467) are also experiencing the same issue, Windows 11 22H2. Can you post more specific instructions on how to grab the debug info you need?

mjansen4857, 2022-10-12:
Follow these instructions: https://github.com/mjansen4857/pathplanner#how-to-build-manually

Make sure you use `flutter run` to run in debug mode. I'm assuming this is an issue with the 22H2 update messing up some dependency needed to run. Building manually could potentially install these dependencies and work fine. If that doesn't work you could try installing Visual Studio Community (make sure the "Desktop Development with C++" module is installed with it) which might fix it as well. If all that fails then I don't think there's much I can do besides wait for a flutter update to fix the issue.

ehamwey, 2022-10-12:
Some initial searching led me to this issue: https://github.com/bitsdojo/bitsdojo_window/issues/190

Working on building from source now.

ehamwey, 2022-10-12:
It looks like Bitsdojo Window library was removed in May, but there hasn't been a release since March. Is it possible to send a newer Windows binary over? The Flutter setup is taking quite a while.

ehamwey, 2022-10-12:
After running the latest version with `flutter run` this confirmed my suspicions. The issue was likely bitsdojo_window. I'll build a new version for Windows and post it here for anyone still having this problem.

mjansen4857, 2022-10-13:
Yeah that sounds right. Here's the one i built that includes some redistributables that some people might not have so its better to use this one. Keep in mind this version might have unknown issues and pathplanner lib doesn't support all of the functionality if you're using the release version.

[PathPlanner-windows.zip](https://github.com/mjansen4857/pathplanner/files/9769335/PathPlanner-windows.zip)

ehamwey, 2022-10-13:
Beat me to it! We'll use this one for now. Thanks for your help.

Gameknight77YT, 2022-10-13:
I have the same problem, here is what I got after running in debug mode.

Launching lib\main.dart on Windows in debug mode...
Building Windows application...
[ERROR:flutter/shell/platform/windows/direct_manipulation.cc(137)] CoCreateInstance(CLSID_DirectManipulationManager, nullptr, CLSCTX_INPROC_SERVER, IID_IDirectManipulationManager, &manager_) failed
Syncing files to device Windows...                                 518ms

Flutter run key commands.
r Hot reload. 🔥🔥🔥
R Hot restart.
h List all available interactive commands.
d Detach (terminate "flutter run" but leave application running).
c Clear the screen
q Quit (terminate the application on the device).

💪 Running with sound null safety 💪

An Observatory debugger and profiler on Windows is available at: http://127.0.0.1:61441/03JnHKQdgtI=/
flutter: ┌───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
flutter: │ RangeError (index): Invalid value: Valid value range is empty: 0
flutter: ├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
flutter: │ #0   List.[] (dart:core-patch/growable_array.dart:264:36)
flutter: │ #1   Trajectory.calculateVelocity (package:pathplanner/services/generator/trajectory.dart:187:13)
flutter: │ #2   Trajectory.generateSingleTrajectory (package:pathplanner/services/generator/trajectory.dart:87:5)
flutter: │ #3   Trajectory.generateFullTrajectory (package:pathplanner/services/generator/trajectory.dart:67:30)
flutter: │ #4   RobotPath.generateTrajectory.<anonymous closure> (package:pathplanner/robot_path/robot_path.dart:79:46)
flutter: │ #5   new Future.<anonymous closure> (dart:async/future.dart:253:37)
flutter: │ #6   _rootRun (dart:async/zone.dart:1383:47)
flutter: │ #7   _CustomZone.run (dart:async/zone.dart:1293:19)
flutter: ├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
flutter: │ 08:10:16.131 (+0:00:01.484156)
flutter: ├┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄┄
flutter: │ ⛔ Dart Error
flutter: └───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
Lost connection to device.

Gameknight77YT, 2022-10-13:
> Yeah that sounds right. Here's the one i built that includes some redistributables that some people might not have so its better to use this one. Keep in mind this version might have unknown issues [...]

This download doesn't work for me either

ehamwey, 2022-10-13:
Just tried to run the same application since yesterday and can't get it to start nor the version I built, but running with `flutter run` in debug mode starts fine.

ehamwey, 2022-10-13:
Can `flutter run` build the app without an Internet connection?

mjansen4857, 2022-10-14:
@Gameknight77YT it looks like one of your path files is messed up for some reason. Try removing them from your robot project and see if it will launch then. 

@ehamwey No idea why it would work in debug mode but not release mode. Only thing I could think of would be to run in debug mode and just don’t close it. I guess worst case you can just connect to a hotspot or Wi-Fi to run it again. Unfortunately flutter checks for dependencies every time you run so you need to be online. There should be a log file in C:/Users/{YOUR USER}/AppData/Roaming/pathplanner something (don’t remember the exact folder) check that after trying to open the built app and maybe it will have something.

Gameknight77YT, 2022-10-14:
> @Gameknight77YT it looks like one of your path files is messed up for some reason. Try removing them from your robot project and see if it will launch then.

I did that and it is working now.  I then readded them one at a time.  The problem was one of them had an endpoint that was also a stop point.

mjansen4857, 2022-10-15:
That probably happened when you deleted another point in the path. I'll make a separate issue to fix that.

mjansen4857, 2022-10-18:
@ehamwey when you get a chance can you pull changes for the repo and see if builds you make will run now? I fixed a crash that *may* have been what caused your builds to crash but I'm not sure. If not then I'm really not sure what's going on there unless you can find some error output.

ehamwey, 2022-10-19:
We had to `flutter pub upgrade` to get the builds working, but once it was built we have an offline version that works. Thanks for your help again. I think I may have moved the Release folder out of the build directory originally which messed with the binary paths.

ehamwey, 2022-10-19:
Looks like after a reboot the built binary does not run. Any thoughts?

mjansen4857, 2022-10-19:
Is there anything in `%AppData%/com.mjansen4857/pathplanner/log.txt`?

ehamwey, 2022-10-19:
Negative - it's empty. 
We were getting an error relating to how the built directories were moved, which seemed to have fixed itself once we connected to WiFi? So I'm trying a clean and build in the final destination directory to see if that resolves things, then we're rebooting and testing offline.

mjansen4857, 2022-10-19:
Moving the built app shouldn't do anything. Just to confirm, after building you're running the exe in `pathplanner/build/windows/runner/Release` right?

ehamwey, 2022-10-19:
Yep - moving the entire `pathplanner` folder. 

Seems like it will run on WiFi, but not with WiFi off.

mjansen4857, 2022-10-19:
By WiFi on/off do you mean connected/disconnected from a network or actually turning the WiFi off in the settings?

mjansen4857, 2022-10-19:
Never mind, I can reproduce it. This is something, thanks. I'll look into it.

ehamwey, 2022-10-19:
I'm going to try on Flutter 3.3.4, we're currently running 3.5.0 because of the method we installed it by. I don't think that should fix the problem but will try anyway.

mjansen4857, 2022-10-19:
Found the issue and fixed it. Was a quirk of how dart try/catch blocks work so checking for PathPlannerLib updates would crash. Pull changes and try again.

I'd definitely use 3.3.4 as well since 3.5 is a beta and I'm currently using 3.3.4.

mjansen4857, 2022-10-19:
So assuming that's fixed, I believe the original issue this thread is based on is a problem with the 2022 version that is not present in the main branch. Here's a pre-release build of the main branch for anyone else finding this issue later: 
[pathplanner-windows.zip](https://github.com/mjansen4857/pathplanner/files/9816023/pathplanner-windows.zip)

I'll close this after the 2023 version is released.

ehamwey, 2022-10-19:
Will try this and report back.

ehamwey, 2022-10-19:
That did the trick (hopefully for good!)

mjansen4857, 2022-10-25:
Beta is released now so I'll close this

closed 2022-10-25 by mjansen4857