mjansen4857/pathplanner #118: Use latest PathPlannerLib in robot dev

tervay, 2022-09-30:
Hi, we're looking to use PathPlanner on an offseason bot, however some specific parts we're looking to use are currently only on the `main` branch, and not on `2022.2.1`.

Is there a recommended way to pull the latest PathPlannerLib into a project?

mjansen4857, 2022-10-01:
A couple options:
* Copy the code for the language you want to use into your project. This is probably the quickest way to do it. 
* Or build from source then copy the built library into the WPILib maven cache. I can give you instructions on doing this if you’d like but it should be a lot easier to just copy the source.

tervay, 2022-10-06:
That will work for now. Any ETAs on a new official release?

mjansen4857, 2022-10-06:
There’s still some stuff I want to get in so not too sure yet. Need to at least wait for a 2023 WPILib beta so I can make any necessary updates for that.

closed 2022-10-06 by tervay