mjansen4857/pathplanner #119: Add documentation for how to use features

isaec, 2022-10-10:
The pathplanner readme advertises the following features:
> - Allows placing "event markers" along the path which can be used to trigger other code while path following.
> - Split a path into a "path group" to follow each part of a path seperately.
Are these still pathplanner features? How can they be triggered or used?

It would be nice to update the documentation to explain how to use them, or update the readme to note they are no longer supported.

I see in the Java library support for accessing the waypoints, but I don't see where they would be generated in the GUI.

It would also be nice to document the ``getConstraintsFromPath`` and on the fly ``generatePath`` methods, even if they feature warnings, so more users are aware of them.

mjansen4857, 2022-10-10:
That’s all new stuff that hasn’t been officially released yet so it’s still undocumented. I will update the wiki before it’s released.

isaec, 2022-10-10:
Is there an ETA or interest / need for contribution? 5026 is interested in using these features.

mjansen4857, 2022-10-10:
I'm mainly waiting for a 2023 WPILib beta at this point. There's a couple other things in the works but nothing preventing a release barring any issues with the beta. Building the app from source is fairly easy if you're interested in getting a head start.

mjansen4857, 2022-10-24:
Ok, I believe all the new stuff is documented on the wiki now. Aiming for a beta release in the next few days.

closed 2022-10-24 by mjansen4857