mjansen4857/pathplanner #128: C++ version PPSwerveControllerCommand release date?

WarrenReynolds, 2022-10-24:
Hi,
Is it likely a C++ version of your PPSwerveControllerCommand will be released in the near future?  Last Season we modified the WPILib SwerveControllerCommand example code to process PathPlanner paths (that included holonomic rotations) instead of the standard WPILib Paths and we also modified the PP code to read the JSON Vel and Acc values form the paths files directly instead of using hard coded max Vel and Max Acc arguments in the constructor, and we eventually got it working really really well, but we were so far out of our depth having to copy and modify library code and we also had some seriously ugly nested sequential and parallel Command Groups happening in order to do things along the path.  

Path Planner has come so far in the last 12 months with all the new features. Thankyou to all the contributors to this project.

We'd love to be able to use the new marker feature to launch commands along the way and use the new path groups feature so all our paths for an auto will all be in the one file. But with a C++ implementation of PPSwerveControllerCommand not released, I'm wondering if I need to go back down the worm hole and get my head back into that space (PTSD coming on) to write our own updated C++ version of SwerveControllerCommand that handles the markers, stop points and array of pointers to commands or should I just hold tight as the C++ implementation is just around the corner?
Cheers
Warren

mjansen4857, 2022-10-24:
The reason it currently isn't there is the vendordep template provided by WPILib does not include the C++ commands library. I have no idea how to add this dependency to the project since it seems to be undocumented and the way the gradle config works is different from a normal robot code project. It seems like it should be added in [here](https://github.com/mjansen4857/pathplanner/blob/main/pathplannerlib/config.gradle#L8) but any name I could possibly think of for it doesn't work. If you have any ideas I'll give them a try.

mjansen4857, 2022-10-24:
I suppose worst case, if I can't figure out how to include it in the library, I can provide source code for the C++ commands that you can just copy somewhere in your project instead of every C++ team needing to write a custom version themselves.

WarrenReynolds, 2022-10-25:
I'm not really up on all the Gradle stuff. I'm more of a, look at sample code and modify it to make it work than fully understanding whats going on under the bonnet type of coder, so I can't offer much help in these types of issues.

Was the command based framework stuff setup as a dependency when they were transitioning from the old command based framework to the new command based framework a couple of years back, so you had a method choosing which command based framework you wanted to use? 

I cant see wpilibOldCommands anymore in the repo. Has the old command based framework now been completely depreciated and the new framework in just part of the core WPILib code and hence it doesn't need to to be optionally selected now?  

All this stuff is way above my understanding, Sorry I can't offer more help.

mjansen4857, 2022-10-25:
Yeah I don’t use gradle outside of FRC so I don’t get much of it either. I’m not really sure if this is how the WPILib compilation stuff works, but I believe that pretty much everything is a shared library, so including a dependency on the new commands framework might mess things up for teams not using a command based project since they won’t have that dependency in their project (although I’m not sure there’s many, if any, teams doing path following without the command framework). I guess I’ll just keep messing around with it for a bit to see if I can figure it out. If not I will post source files somewhere for c++ teams before the season starts.

WarrenReynolds, 2022-10-25:
We need someone that works on WPILib command based stuff to help out here. I think oblarg from the Wpilib Repo is the person that re-wrote the New C++ Command Based framework.  What is the exact question we need to ask him?

mjansen4857, 2022-10-25:
I don’t think it’s anything to do specifically with the commands framework since it’s typically included as a vendordep but the vendordep template needs it to be included differently by editing the gradle files somehow. I’ll probably just open an issue in the vendor template repo to ask what I need to change.

MrRedness, 2022-10-25:
> Yeah I don’t use gradle outside of FRC so I don’t get much of it either. I’m not really sure if this is how the WPILib compilation stuff works, but I believe that pretty much everything is a shared [...]

MrRedness, 2022-10-25:
> Although I’m not sure there’s many, if any, teams doing path following without the command framework
It's not as bad as you might think
In fact, I prefer knowing what is actually happening on each loop than abstracting it into a command

MrRedness, 2022-10-25:
Very interesting stuff @WarrenReynolds , I haven't considered the scalability of commands the way you phrase it. Our robot did most of the same stuff during autonomous and teleop with the timed framework. While I admit many of things we do would be considerably easier with commands, I still believe the flexibility of timed makes up for it. I prefer to spend more time writing something that I am in full control of rather than going with an easier solution and having to work around it if necessary. As far as scalability/readability goes, I'd say our codebase is quite good. Most, if not all, functions are written with the either the intention of being run one or looped, with some including loops and threads themselves. If you go into it with that mentality, almost everything is intuitive. Our codebase is spread across 19 cpp files and 29 hpp files. Here's an example of the buttonManager() function, which we call in teleopinit to handle our cargo path. 

[image]

I definitely would like to do a further exploration into command based framework, especially as C++ support becomes more flushed out, but for now we're happy with what we use.

mjansen4857, 2022-10-26:
If you're interested in playing around with it, I made a working [C++ version of PPSwerveControllerCommand](https://gist.github.com/mjansen4857/19f4a33bf0b3cc930b9c632c835f7504). Its just a single .h file since I couldn't get it to compile as a .h and .cpp file for some reason. No idea why. I suppose if you want it separated you can try to separate it yourself and see if you have more luck. It seems like depending on the command framework is not really going to be an option so I will make more gists for the other commands at some point and document using them on the wiki.

mjansen4857, 2022-11-19:
Someone over in vendor template repo suggested a change that will let me depend on the new commands vendordep so these commands will now be added directly to PPLib. Downside being that teams not using the command based vendordep will probably need to install it. Unsure about if it will care if you never actually use any of the commands.

closed 2022-11-19 by mjansen4857