mjansen4857/pathplanner #130: Ability to add a "stop time" for each marker/stop point

mjansen4857, 2022-10-25:
### Discussed in https://github.com/mjansen4857/pathplanner/discussions/129

<div type='discussions-op-text'>

<sup>Originally posted by **MrRedness** October 25, 2022</sup>
This way, you could change how long the robot waits at a marker/stop point without having to change it in code and redeploy. I would be happy to work on this when I have a little bit more time.</div>

WarrenReynolds, 2022-10-26:
Yeh, I must admin I hadn't thought through exactly how I would implement what I was talking about and I can see the problems you are talking about with the follower not having any knowledge of the commands that it needs to execute.  Could you create an array of pointers to commands that you are going to use while following the trajectory in your RobotInit or AutoInit and then pass this object into a modified version of your PP follower code that take this as an optional extra argument?

Create the array as a map object in this format
map<datatype_of_keys, datatype_of_values>name_of_the_map
So start with this line, maybe somewhere in the RobotContainer class.
         map<string,*command>m_ListOfCommands;
Then for each command you are wanting to use at the various markers add an entry to your map.
        m_ListOfCommand["OpenIntake"]= new OpenIntake());  //Where OpenIntake() is the command
Then when you execute your trajectory, have a constructor on the trajectory follower that takes an optional extra argument so that you can pass this map of command pointers into the follower.

Now your follower has the ability find a command by name (by iterating through the dictionary of pointers to commands until it finds a matching name to the value pair of the CmdToExecute pair in the path) and then dereference this command pointer to add the command to the scheduler.

I admit this is not very nice code wise but it does give a solution to accessing commands by name without a heap of if statements inside the follower.
If the map of pointers to commands wasn't going to be used again after you have finished following your trajectory, you should probably delete all the instances of the heap allocated commands after finishing the trajectory to release the memory. Or you might keep this map if you ever want the ability to launch a command by name somewhere else in your code.   

I just re read what you said above.  I'm not sure what you were referring to when you are talking about an EventMap. Maybe we are on the same page already.

Much food for thought here.

WarrenReynolds, 2022-10-26:
Probably a dam lot easier to just create a straight array of pointers to commands and then use the marker number to find the entry in the array of pointers as a straight array index lookup instead of having to look up text names of commands in dictionary objects.

Although if you were to do it this way and you wanted to open the intake at say marker 3 and marker 7, this would require to two instances of the openIntake() command to be stored in your array of command pointers.  Actually, no it doesn't, you could just reuse the same pointer value at position 3 and 7 in the array of pointers.  

In all honesty I really have no idea what the best way would be to do this, but I think this is how I'll implement it, if the functionality isn't there by the 2023 season.  This could be implemented with only a very slight change to the path follower, and it doesn't need any changes to the trajectories markers as they stand at the moment to make all this work.  PS: I'd love a copy gof  your C++ version of the follower as it stands at the moment, so I can make our custom version.

NB: Could get a bit painfully if you insert a new marker onto your path, It means you would need to change the code where the pointers to commands are created so that all commands after the newly inserted command would need their position in the array moved down by one.  

Maybe adding key,value pairs to markers of type <string>,<int> is all that is needed and then let people do what they want with the key values.  If they don't exist, it doesn't matter, if they do exist the end user is responsible to know what the key name is and what to do with the value that is associated with it.  This adds all the functionality we have been talking about and leaves the implementation of the information in these key value pairs up to the end user.

mjansen4857, 2022-10-27:
By "event map" i mean the map from marker names to command to run that is passed to the path following commands. I wanted to avoid arrays because if you change the order of markers or add/remove them you will need to edit code like you mentioned. My idea of how markers would be best used was to just create a "global" map of marker names to commands and pass this map to every path following command you run. So regardless of what path or auto you run "intakeDown" will always run the same thing. If you build up this map with all the commands you would run in an auto you can freely edit the markers for a path and not need to touch any code to add/remove/change commands in an auto (the ones using markers at least).

WarrenReynolds, 2022-10-27:
Sounds perfect.

closed 2022-10-27 by mjansen4857