mjansen4857/pathplanner #132: PathPlannerServer crash

CrispyBacon1999, 2022-10-26:
We just redesigned our autonomous routines with the new beta release and everything was running pretty smoothly. We had PathPlannerServer running on it, which worked great, until it crashed, causing the robot to continue driving for a second or two. We'd love to be able to use this for development. Let me know if you need any more information about this.

Crash report:
```
6:39:05.447 PM
   	at frc.robot.Main.main(Main.java:18)  	at edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:463)
  	at edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:373)
  	at edu.wpi.first.wpilibj.TimedRobot.startCompetition(TimedRobot.java:131)
  	at edu.wpi.first.wpilibj.IterativeRobotBase.loopFunc(IterativeRobotBase.java:328)
  	at frc.robot.Robot.robotPeriodic(Robot.java:59)
  	at edu.wpi.first.wpilibj2.command.CommandScheduler.run(CommandScheduler.java:280) 
 	at edu.wpi.first.wpilibj2.command.SequentialCommandGroup.execute(SequentialCommandGroup.java:67)
  	at edu.wpi.first.wpilibj2.command.ParallelDeadlineGroup.execute(ParallelDeadlineGroup.java:93)
  	at edu.wpi.first.wpilibj2.command.SequentialCommandGroup.execute(SequentialCommandGroup.java:67)
  	at edu.wpi.first.wpilibj2.command.SequentialCommandGroup.execute(SequentialCommandGroup.java:67)
  	at com.pathplanner.lib.commands.PPSwerveControllerCommand.execute(PPSwerveControllerCommand.java:139) 
 	at com.pathplanner.lib.server.PathPlannerServer.sendPathFollowingData(PathPlannerServer.java:122)
  	at com.pathplanner.lib.server.PathPlannerServer.sendToClients(PathPlannerServer.java:46)
  	at java.base/java.util.ArrayList$Itr.next(Unknown Source)
  	at java.base/java.util.ArrayList$Itr.checkForComodification(Unknown Source)
Error at java.base/java.util.ArrayList$Itr.checkForComodification(Unknown Source): Unhandled exception: java.util.ConcurrentModificationException
ERROR  1  Unhandled exception: java.util.ConcurrentModificationException java.base/java.util.ArrayList$Itr.checkForComodification(Unknown Source) 
```

mjansen4857, 2022-10-26:
Thanks for the report, I’ll look into it.

mjansen4857, 2022-10-26:
Are you able to reproduce this a lot or was it a one time thing?

mjansen4857, 2022-10-26:
I may have pushed a fix for this but it's hard to know since the issue I think it is should be very rare and I couldn't get it to happen. If it does happen a lot for you let me know because it's probably a different issue then. Otherwise, I guess wait to see if it happens again whenever PPLib gets updated.

CrispyBacon1999, 2022-10-26:
We only had it once before disabling it to prevent damage to the robot, we can definitely test it out some more to check if it's still an issue.

mjansen4857, 2022-10-26:
Well if you try it again maybe just ignore the output from the command so you don’t break anything. It should only be able to happen after the GUI loses connection with the server so closing the app while it’s the robot is path following could be a way to trigger it.

mjansen4857, 2022-10-27:
Nevermind, I just got the error myself so the fix didn't work. Will keep looking

mjansen4857, 2022-10-27:
Ok I found a way to reproduce it and it seems to actually be fixed now so I'll close this. I guess just leave the server disabled for now until the next update.

closed 2022-10-27 by mjansen4857