mjansen4857/pathplanner #133: Angular velocity is not constrained by max acceleration and velocity

laviRZ, 2022-10-26:
When making a path with high curvature, I see the angular velocity going above 7, even when I limit it to 1.

mjansen4857, 2022-10-26:
Max acceleration and velocity are constraints on linear velocity/accel, not angular. Angular velocity is also radians/second, not meters/second so they are different  Angular velocity is calculated based on the robot's linear velocity and curvature for differential drives. If the angular velocity is something you would consider too high then you need to reduce the curvature by smoothing the path out.

Going to close this as it's functioning as expected.

closed 2022-10-26 by mjansen4857

laviRZ, 2022-10-26:
Maybe set a limit that represents the maximum velocity for a module? Then you can calculate off of that the linear and angular limits

mjansen4857, 2022-10-26:
If you're talking about a holonomic drive train then angular velocity is not related to curvature and is just based on how much the robot needs to rotate to hit the holonomic rotation set at the next waypoint. You need to set your holonomic rotations with the fact that the generation does not account for them in mind. Basically you need to make sure the rotation is something the robot can actually accomplish.

> Maybe set a limit that represents the maximum velocity for a module?

To support stuff like this would be nice, but would be a very significant change to the generation, which is not something that I would be comfortable doing for the 2023 season at this point, since there's not much time to do it right and test it out. There are other generation improvements I have in mind that I want to make, this being one of them, that I would like to look into with a generation re-write for next year.