mjansen4857/pathplanner #139: Multiple markers at a point

lidor51, 2022-10-29:
Being able to choose a point that will trigger multiple markets at once.

E.g., Assuming we already support "Markers list/dropdown", and having auto routine with two places we want to trigger subsystems commands, at the first point "Open Intake" & "Accelerate Shooter", second is "Close Intake" & "Accelerate Shooter".

Assuming we already have those 3 markers in our dropdown an use them in different routines in our project.
As for now we would need to add "Open Intake and Accelerate Shooter" and "Close Intake and Accelerate Shooter" unified markers, and corresponding two Sequential Commands (or list of command as in your suggestion in CD). You can assume that after writing all routines, you'll find yourself with so many very specific markers and sequential commands.

Instead, my suggestion is just letting in the marker editor add multiple markers, each can reflect a simple command in the robot project.

mjansen4857, 2022-10-29:
Changed title to be a bit more clear.

WarrenReynolds, 2022-10-30:
This would be a nice feature but I think we need to give Michael a break here.  He has done so much getting his program to where it is already and we are asking for further changes (before I think a lot of us have even tried the current version).  Sure it would be a lovely feature to have, but I feel firing a predefined command group does achieve the same result.  After all you can define a command group inline with a single line of code in your robot container.  

Also would these multiple commands at a marker point be sequential or parallel or a mixture of both, do you then need to add more functionality to PP so that sequential and parallel commands are handled in this new marker point?

If this feature was to be incorporated, to save another re-write of the underlying data structure of a marker point to have multiple names, would it be easier to have the ability to add multiple markers at the exact same location.  (ie Duplicate a marker with the same position setting)

I know the PP UI might need a bit of a re-write in order to display all the markers.  Maybe a different coloured marker to show there are multiple markers on top of each other, and then a couple of navigation arrows on the edit marker dialog box to advance forward or backwards through the list markers, so you have the ability to edit the names of the markers that are stacked on top of each other.

Or another option is put this type of customisation back onto the teams and get them to customise their followers to deal with this.  A simple solution would be to use a single marker name that is comma separate string of the multiple command names you want to fire at this marker.  Then in the follower code, the team needs to pull apart this string to get the list of command to fire. You could even add things like "S:Command1,P:Command2,P:Command3". Then the follower code could then pull this string apart and build a command group that sequentially runs Command1 and then Command 2 and 3 in parallel.  Actually I'm not sure if this is even possible in the command base framework. Can a command group be dynamically created at run time? I don't think it can be. Scratch adding sequential commands that's getting to hard.   It has to be parallel only if this is going to be  implemented.

mjansen4857, 2022-11-06:
> Also would these multiple commands at a marker point be sequential or parallel or a mixture of both, do you then need to add more functionality to PP so that sequential and parallel commands are handled in this new marker point?

The multiple events would just all be scheduled at once. It's up to the user to make sure their commands wont interrupt each other. If you need to do something in sequence, then you'll just have to add a command group to the event map.

> I know the PP UI might need a bit of a re-write in order to display all the markers. Maybe a different coloured marker to show there are multiple markers on top of each other, and then a couple of [...]

It's just a list of names in the marker instead of one name. So multiple text boxes/dropdowns (eventually) on one marker.

closed 2022-11-06 by mjansen4857