mjansen4857/pathplanner #152: Build issue with PathPlannerLib 2023.0.1

THRYLLR, 2022-11-19:
I have implemented a WPILib Ramsete command in C++, and I used pathplannerlib's `PathPlanner::LoadPath` to retrieve the trajectory we designed in PathPlanner. Upon building this code, I get a large linker error which causes the build to fail:

```
> Task :linkFrcUserProgramReleaseExecutable FAILED
C:\Users\<me>\.gradle\caches\transforms-3\81738442bbe4daeb1060f4f22a2925e9\transformed\PathplannerLib-cpp-2023.0.1-linuxathenastatic\linux\athena\static\libPathplannerLib.a(PathPlanner.o): In function `pathplanner::PathPlanner::generatePath(pathplanner::PathConstraints, bool, pathplanner::PathPoint, pathplanner::PathPoint, std::initializer_list<pathplanner::PathPoint>)':
PathPlanner.cpp:(.text+0x7c8): undefined reference to `std::__throw_bad_array_new_length()'
C:\Users\<me>\.gradle\caches\transforms-3\81738442bbe4daeb1060f4f22a2925e9\transformed\PathplannerLib-cpp-2023.0.1-linuxathenastatic\linux\athena\static\libPathplannerLib.a(PathPlanner.o): In function `pathplanner::PathPlanner::loadPath(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, pathplanner::PathConstraints, bool)':
PathPlanner.cpp:(.text+0x1dc8): undefined reference to `std::__throw_bad_array_new_length()'
C:\Users\<me>\.gradle\caches\transforms-3\81738442bbe4daeb1060f4f22a2925e9\transformed\PathplannerLib-cpp-2023.0.1-linuxathenastatic\linux\athena\static\libPathplannerLib.a(PathPlanner.o): In function `pathplanner::PathPlanner::loadPathGroup(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::initializer_list<pathplanner::PathConstraints>, bool)':
PathPlanner.cpp:(.text+0x31f8): undefined reference to `std::__throw_bad_array_new_length()'
PathPlanner.cpp:(.text+0x3364): undefined reference to `std::__throw_bad_array_new_length()'
PathPlanner.cpp:(.text+0x3370): undefined reference to `std::__throw_bad_array_new_length()'
C:\Users\<me>\.gradle\caches\transforms-3\81738442bbe4daeb1060f4f22a2925e9\transformed\PathplannerLib-cpp-2023.0.1-linuxathenastatic\linux\athena\static\libPathplannerLib.a(PathPlanner.o):PathPlanner.cpp:(.text+0x33ec): more undefined references to `std::__throw_bad_array_new_length()' follow
collect2.exe: error: ld returned 1 exit status
```

I highly doubt that this is due to an error with my code as I can fix this error and have a successful build by downgrading to version 2022.2.1. As there is no native mechanism I am aware of to DOWNgrade vendor dependencies, I accomplished this by copying the vendordep .json file for 2022.2.1 from an older project into this one and rebuilding.

mjansen4857, 2022-11-19:
Could you share the code for your project? Are you using the 2023 beta?

THRYLLR, 2022-11-19:
I'm using the latest version of the app available on the Microsoft Store (v2022.2.1 I believe), and I'm using the latest version of pathplannerlib at the URL [](https://3015rangerrobotics.github.io/pathplannerlib/PathplannerLib.json), which is in fact the 2023.0.1 beta. I'm not sure why the vendordep for a beta version is located at this URL which is present in the docs for a stable version.

This is part of our `GetAutonomousCommand`:

```cpp
frc2::Command* RobotContainer::GetAutonomousCommand() {
  pathplanner::PathPlannerTrajectory autoTrajectory = pathplanner::PathPlanner::loadPath(TRAJECTORY_NAME, DriveConstants::kMaxAutoSpeed, DriveConstants::kMaxAutoAccel);
  frc::Trajectory WPItrajectory = autoTrajectory.asWPILibTrajectory();

  m_drivetrain.ResetOdometry(WPItrajectory.InitialPose(), WPItrajectory.InitialPose().Rotation());

  frc2::RamseteCommand followTrajectory {
    autoTrajectory.asWPILibTrajectory(),
    ...
```

mjansen4857, 2022-11-19:
I meant WPILib beta. If you're still using 2022 WPILib this is likely the problem.

THRYLLR, 2022-11-19:
Oh right, I am not using the WPILib beta right now; I'm using the latest 2022 release. That must be the issue. In any case, would it be possible to put a different, stable vendor dependency URL on the wiki alongside the beta one to avoid potential confusion?

mjansen4857, 2022-11-19:
I suppose, but the only time there is ever a "beta" release is the pre-season. After kickoff there will be no beta releases at all, everything will be a "stable" version. Since the wiki is always updated for the latest version, this "stable" version of the vendor dep could lead to some confusion for others in the pre-season if a vendordep URL is provided on a wiki describing some feature that isn't actually present in that version. I think its most likely better to just have the single URL, as anyone using pathplanner in the offseason is likely at an offseason event, and not updating pplib, or trying the beta release. Sure, there's a few people doing neither of those things, but its such a small number it doesn't really make sense to spend the time maintaining separate documentation and maven releases.

closed 2022-11-19 by THRYLLR

MrRedness, 2022-11-19:
> As there is no native mechanism I am aware of to DOWNgrade vendor dependencies, I accomplished this by copying the vendordep .json file for 2022.2.1 from an older project into this one and rebuilding.

For future note, you can manually downgrade just by editing the version number in the .json file of the dependency (located in the vendordeps folder). This is best done with a control-H to replace all references of the 'new' version (in this case 2023.0.1) with the 'old' version (2022.1.1). This also works for manual upgrading.