mjansen4857/pathplanner #156: Stop Point Event List won't populate

WarrenReynolds, 2022-11-21:
This stop Point event list functional looks amazing, but unfortunately current version doesn't allow you to save event names at stop points, or populate more than one entry.

WarrenReynolds, 2022-11-21:
Just updated from Flutter 3.3.1 to Flutter 3.3.8 and rebuilt.  Stop Point commands are now being added successfully. Will close now.

closed 2022-11-21 by WarrenReynolds

mjansen4857, 2022-11-21:
If you’re just trying to run the main branch before it’s released instead of contributing, the CI is now set up to build the app after every commit to the main branch. Downloading a guild from there may be easier than building manually. Just click on the Actions tab and you should find it

WarrenReynolds, 2022-11-21:
Sorry, I'm not very good at this github stuff and I'm learning something new everyday.  I've never used Actions before.
What is the CI?
What is a guild and how do I download it?  Is this the link to the exe that is created for windows?

And sorry that I'm not contributing, I would if I had something to offer.

mjansen4857, 2022-11-21:
CI stands for "Continuous Integration". Basically, it just means that every time a commit is pushed to main or a pull request is made, the app is automatically built and tested. I meant "build" not "guild". Typo.

Here's a link to the workflow that builds the GUI app : https://github.com/mjansen4857/pathplanner/actions/workflows/pathplanner-ci.yaml

Each entry in their is either a commit or a pull request. If you click on the most recent one for example, downloads for the builds will be at the bottom:
[image]

The "MSIX" one is the file needed to upload to the windows store, so you can ignore that one.

Also, there's no need for anyone to contribute at all lol, don't be sorry.

WarrenReynolds, 2022-11-22:
Yep, Got it now.  I'd clicked on the first green tick I saw, which I can now see is the workflow for building PathPlannerLib, not 
PathPlanner.  Completely makes sense now, when I click on the right thing.

Thanks again for everything you do in this space.  I think you have nailed it completely in terms of the functional of your program. That's a massive achievement.  I've certainly learnt a lot about c++ command based coding following the evolution of your program.  I hope to learn about flutter and dart in the future so I can contribute to Pathplanner outputting trajectory files for FLL. 
If we make it over to Houston next year for the champs, I'll track you down and buy you some coffees, I certainly owe you.

mjansen4857, 2022-11-22:
You don't owe me anything lol. Helping others out is just part of being involved with FIRST. And I actually don't drink coffee anyways so there's no need for that lol.

WarrenReynolds, 2022-11-22:
Quick question, where do all the Global setting for PathPlanner live (registry or a setting file)?  It would be nice to be able to switch between FLL mode and FRC Mode. Each have a different field image, different field size/pixel ratio, different robot sizes, different projects folders.

If all this stuff lives in one file or reg key that I could copy/rename to switch between modes this would be easy way for me to change modes. It's a bit painful to do this at the moment. BTW: I'm not expecting this to be a feature, as I'm probably the only person in world that using your program this way.

mjansen4857, 2022-11-22:
I think you're on windows right? If so, `%AppData%\com.mjansen4857\pathplanner\shared_preferences.json`

WarrenReynolds, 2022-11-22:
Yep windows. Perfect, exactly what I was looking for.  

And I can tell you already thinking about it, do I add a backup/restore shared preference button in the setting tab or do I make the exe take an optional command line argument that ignores the default setting files and loads a custom setting from the file supplied.

mjansen4857, 2022-11-22:
I doubt flutter supports command line arguments.