mjansen4857/pathplanner #159: Add "At least" wait behavior

lidor51, 2022-11-22:
At Least wait behavior will run the wait and the events in Parallel Command Group, which will cause to robot to immediately execute events, but delay the start of next path at least by the defined wait time.

mjansen4857, 2022-11-22:
This is very similar to how parallel deadline group works. The only difference being that parallel deadline will cancel running commands after the deadline and this "at least" method would not.

Do you have an example of where this would be useful? Just trying to have a use case in mind.

lidor51, 2022-11-22:
Yep I'm aware to this subtle difference, thus suggested to add this as well.

A made up example use case, you want to turn to angle but you also don't want to start the next path before X seconds will pass from when you finished the last path  (because you believe other robot might pass in that time frame for example).
You can use workarounds with your current tools, but it's easy enough to add this option. Sure that teams will be creative with it.

WarrenReynolds, 2022-11-22:
To achieve this behaviour, wouldn't you just add a delay command to the list of commands to run in parallel.

mjansen4857, 2022-11-22:
From the auto builder side, yeah. But it makes sense to add a GUI option to do this for you since if you just did this with the existing behaviors you may need to add "wait0.5", "wait1.0", "wait1.5" etc to the event map to use different timing. Or have to change code if you want to adjust timing. Being able to use the wait time setting for this makes it easier.

WarrenReynolds, 2022-11-22:
Agreed. I just edited my post.

mjansen4857, 2022-11-22:
I think in the future this can just be replaced by letting you add configurable wait commands anywhere. But that's further down the line with a GUI re-work focused on auto builder stuff.

WarrenReynolds, 2022-11-22:
You could add some text parsing to the marker names, If the marker is called "AutoAim", the runner would run the command associated with AutoAim in the event map.  If the marker name was called "AutoAim,7000", the runner could break apart this string and search the event map for AutoAim and then add the command the group with a parameter/argument of 7000.  An example case for this might be a shooter speed that could be defaulted if the lime light didn't detect the target. I thinking an integer would be the most likely form of this parameter/argument.  If team wanted to pass a float like 3.5 for a time delay they should make their command take milliseconds and pass 3500 as the parameter/argument.  I think this would just about cover most custom use cases and it wouldn't be that hard to implement.

lidor51, 2022-11-22:
Warren, I wouldn't advice my team to use a feature like that. Just not so aesthetic.

And also increase the chances for runtime errors (or what if you need to parse complex objects as params).

WarrenReynolds, 2022-11-22:
I'm just suggesting a solution that would cover most use cases as 99% of commands could take integers as arguments.  To pass custom objects to commands from inside path planner GUI is something that I don't thing will ever be handled.  The UI would get some messy to enable this.

It's only an idea, and I'm thinking our team will run with this approach with a bit of custom code in the auto followers as it can be implemented in only a couple of lines code and we don't have to touch the path planner GUI to make the changes.

mjansen4857, 2022-11-22:
I don't think that's even possible. PPLib has no way of knowing what your commands are or how to construct them. You can't pass an unconstructed object to the event map that pplib could just pass a parameter to. The only way to do something like this would be to do it in a custom auto builder.

WarrenReynolds, 2022-11-22:
Yep, good point.  The command has already been created beforehand and you only have a pointer to the command in the command map.  

Could this work as a workaround to pass values to a command that has already been constructed.

Create a new class of command (Called PPCommand) that inherits everything a normal FRC command has, but also includes a special setter function (eg  PPParameterSetter(int) ) that you override that allows you to get an integer into a member variable inside the command. (Just like how the constructor code of a command usually sets the commands member variables) 

Then the follower would match the marker name with the names in the command map, get the pointer to the command, de-reference this pointer and call the special setter function on the command to update the member variables for this instance of the command.

So now your command has is internal member variable set correctly. Then add this command to the command group to be run at the marker or stop point.  Maybe a cast back to a normal FRC command_ptr would be needed before it is scheduled. (Not sure on this)

After re-reading this post, you are right, this is not easy problem to solve.  You almost need a special overloaded class of commands that all commands in the command map must be of this type if you want to be able to adjust these commands at run time.

closed 2022-11-28 by mjansen4857