mjansen4857/pathplanner #163: At Stop Point Allow For Un-smooth Curvature

ohowe1, 2022-11-24:
[image]

Would it make sense for at a stop point to allow for the curvature lines to not be one and to allow for it to be separate on each side of it?

WarrenReynolds, 2022-11-24:
It makes sense for swerve drive, but what about regular tank drive.  You would with have a violent turn on the stop at the start of the 2nd segment because the robot is trying to accelerate the centre point of the robot while at the same time trying to correct the heading.  For tank drive I'm thinking this might cause errors in path following.  Even for swerve drie robots all of your modules are going to be facing the wrong way as the 2nd segment is starting. I know the delay in getting them to face the correct would be very small but again it does introduce inconsistency's

A work around would be to shorten the tangent handle on the 2nd leg and then add another waypoint that is in the direction you want to head in but very close to the stop point.  This pretty much does what you want and also give a tank drive robot a short distance to perform its turn.

ohowe1, 2022-11-25:
> It makes sense for swerve drive, but what about regular tank drive.

When I was thinking about this it was in regard to swerve, so probably would be a holonomic exclusive feature considering your point.

> A work around would be to shorten the tangent handle on the 2nd leg and then add another waypoint that is in the direction you want to head in but very close to the stop point. This pretty much does what you want and also give a tank drive robot a short distance to perform its turn.

Yeah, I'm aware of the workaround, just thought it would be better to not have to work around it.

mjansen4857, 2022-11-25:
The biggest issue I have with this is that paths can be loaded as a single path, ignoring stop points. If someone were to try to do this with a break in the path at the stop point it would generally break the path following. I'm not sure if this is exactly what happens, but I believe in a case like this the generation would enter an infinite loop. It would see the break as a curve with 0 radius, and slow down to 0 to not go into that curve too fast. Once its at 0, it can't make forward progress so it can't speed up and just sort of gets stuck there.

So the options for dealing with this are:
1. Throw an exception if you try to load a path with uncoupled headings at a stop point. This is pretty annoying because if you want to test a path without stopping at stop points you would have to make sure you remove all of these breaks, changing the path you want to test.
2. Make a change that would stop the infinite loop. Doesn't really help much as the robot would either still need to slow down significantly, defeating the purpose of ignoring stop points, or go straight through it and cause a bunch of wheel slip and ruin your accuracy.
3. Make a smooth path out of 1 of the 2 decoupled control points. Then which one do you choose? They could result in wildly different paths depending on their placement, and either way will not be the path that you made.
4. Just let it happen. Then people are going to run into this and not understand what's happening.

I just don't think there's that much of a benefit to decoupled headings to justify any of these drawbacks. You can usually get a smooth path that is pretty close to what heading breaks would do by adjusting your point placements.

Obviously there is a benefit, but I don't think its that significant. I will do some playing around with it to see if I can find an implementation for it that I'm happy with.

PR #317 (2023-08-08): 2024 Reworks
Majority of the reworks for the 2024 release.

Fixes #316, closes #311, closes #304, fixes #301, closes #289, closes #285, closes #274, closes #260, fixes #226, fixes #210, closes #163, closes #138, closes #126

Supersedes #300 

TODO:
- [X] GUI reworks
- [X] Java PPLib reworks
- [X] C++ PPLib reworks (missing pathfinding for now)
- [X] Test Java PPLib in robot project
- [x] Documentation
- [x] Finish GUI unit tests
- [x] Test (and probably fix) C++ PPLib in robot project
closed 2023-08-08 by mjansen4857