mjansen4857/pathplanner #164: StopEvent events not being saved

lidor51, 2022-11-24:
StopEvent's events not being saved at all. Regular markers do work as expected.

The behavior is:
If I enter an event name and press enter, the rectangle stop being highlighted, but no additional "Add new event" rectangle is being added below.

If I change anything (Execution/Wait behavior, wait time) the event is being deleted. Also if I move to another tab/marker and come back, the event is gone.

(Also why it's required to press Enter in order to make things to save?)

lidor51, 2022-11-24:
After I manually edit the .path file in one of the StopEvents, to include in it the events names, it made this StopEvent and all other **already existing** StopEvents to work as expected on this path (but it didn't fix the behavior for StopEvents I add after the manual change, neither on new paths).

WarrenReynolds, 2022-11-24:
I had same issue when building from source with an older version of Flutter.  Once I upgraded flutter everything worked as expectd.

See this closed issue.
https://github.com/mjansen4857/pathplanner/issues/156#issue-1457612769

I also raised an issue with the EnterKey thing.  Apparently this one is a hard one to fix.  It's the built in behaviour of Dart and Flutter.
It's on the to do list
https://github.com/mjansen4857/pathplanner/issues/113#issue-1362606048

lidor51, 2022-11-24:
I installed through Microsoft Store.

WarrenReynolds, 2022-11-24:
As a work around until the Store is updated to the latest build, maybe try installing the latest build for Microsoft Store (the MSIX file) directly. This file is built every time the actions run.  (see issue #156, Michael explained it all to me very well)

I think Michael updates the store manually (on major changes), so it might be an older version on the store (that could have the bug you are referring to). If the latest build for MS Store doesn't have the issue you are referring to, then at least Micheal will know. 
Unfortunately I'm on a school managed device so I can't test this theory as the MS Store is locked down for me.

mjansen4857, 2022-11-24:
Windows store is updated. You can actually install the msix file since it’s a store build. 

I think I’ve seen this issue before while debugging and if it’s caused by the same thing I saw then it’s a flutter error. Not really anything I can do about it until flutter is updated. When it happens you just need to restart the app and it should work u til it happens again.

WarrenReynolds, 2022-11-24:
Sorry,

I just doubled clicked on the msix file directly and it opened an installer that looked like it would install. 
[image]

 I thought it wouldn't let me install because of our organisation GP setting.  My bad.

mjansen4857, 2022-11-24:
Yeah. MSIX files can only be installed if they are signed. Store builds are unsigned since the store handles the signing.

WarrenReynolds, 2022-11-24:
Yet again, I learn something new from you.
Cheers

mjansen4857, 2022-11-29:
I haven't seen this since the update to flutter 3.3.9 (even though it doesn't seem to have a fix that would affect this). Can you try the latest windows build from [here](https://github.com/mjansen4857/pathplanner/actions/workflows/pathplanner-ci.yaml) and see if it still happens for you?

lidor51, 2022-11-30:
I ran this version https://github.com/mjansen4857/pathplanner/actions/runs/3570997295, still happens.

mjansen4857, 2022-11-30:
Can you upload your log file: `%AppData%\com.mjansen4857\pathplanner\log.txt`

lidor51, 2022-12-01:
[log.txt](https://github.com/mjansen4857/pathplanner/files/10131295/log.txt)

mjansen4857, 2022-12-01:
Ok so this was a different issue entirely from what I was seeing then. It should be fixed now. Can you confirm by trying the build from [here](https://github.com/mjansen4857/pathplanner/actions/runs/3595115218)

lidor51, 2022-12-01:
Cool, fixed,

closed 2022-12-01 by lidor51