mjansen4857/pathplanner #165: Add a way to delete custom fields or edit their pixels to meters ratio

Teddy-Mengistie, 2022-11-27:
I am an average programmer and cannot deal with dimensions so I created a bunch of custom fields until the pixel to meter ratio was about right. Sadly I cannot delete all of the fields I created :(

Also, to avoid making a bunch of fields in the first place, an "edit field parameters" option would be very nice.

mjansen4857, 2022-11-27:
Wanted to do this for a while just needs some annoying changes to the settings UI so I've been putting it off for a bit lol. Will definitely do this soon though. In the meantime, you can delete unwanted field images from `%AppData%\com.mjansen4857\pathplanner\custom_fields` assuming you're on windows.

Teddy-Mengistie, 2022-11-27:
Worked, ty!

closed 2022-11-29 by mjansen4857