mjansen4857/pathplanner #173: RamseteAutoBuilder Constructor Error

kbyquist, 2022-12-08:
When constructing a RamseteAutoBuilder, the constructor is having errors converting types

```RamseteAutoBuilder autoBuilder = new RamseteAutoBuilder(
    m_robotDrive::getPose,
    m_robotDrive::resetOdometry,
    new RamseteController(),
    m_robotDrive.kinematics,
    new SimpleMotorFeedforward(DriveConstants.ksVolts, DriveConstants.kvVoltSecondsPerMeter, DriveConstants.kaVoltSecondsSquaredPerMeter),
    m_robotDrive.getWheelSpeeds(),
    new PIDConstants(DriveConstants.kPDriveVel, 0, 0),
    new PIDConstants(DriveConstants.kPDriveVel, 0, 0),
    m_robotDrive::tankDriveVolts,
    SystemConstants.eventMap,
    m_robotDrive
    );
```

results in the following error when building robot code:

```
Users/kaseybyquist/Desktop/Robotics/2023 Preseason/PathPlannerBeta/src/main/java/frc/robot/RobotContainer.java:42: error: no suitable constructor found for RamseteAutoBuilder(m_robotDri[...]tPose,m_robotDri[...]metry,RamseteController,DifferentialDriveKinematics,SimpleMotorFeedforward,DifferentialDriveWheelSpeeds,PIDConstants,PIDConstants,m_robotDri[...]Volts,HashMap<String,Command>,DriveSubsystem)
  RamseteAutoBuilder autoBuilder = new RamseteAutoBuilder(
                                   ^
    constructor RamseteAutoBuilder.RamseteAutoBuilder(Supplier<Pose2d>,Consumer<Pose2d>,RamseteController,DifferentialDriveKinematics,SimpleMotorFeedforward,Supplier<DifferentialDriveWheelSpeeds>,PIDConstants,BiConsumer<Double,Double>,HashMap<String,Command>,Subsystem...) is not applicable
      (argument mismatch; DifferentialDriveWheelSpeeds cannot be converted to Supplier<DifferentialDriveWheelSpeeds>)
    constructor RamseteAutoBuilder.RamseteAutoBuilder(Supplier<Pose2d>,Consumer<Pose2d>,RamseteController,DifferentialDriveKinematics,BiConsumer<Double,Double>,HashMap<String,Command>,Subsystem...) is not applicable
      (argument mismatch; SimpleMotorFeedforward cannot be converted to BiConsumer<Double,Double>)
1 error
```

hadley31, 2022-12-09:
This is not an issue with the library. The error says that you’re passing in a `DifferentialDriveWheelSpeeds` instead of a `Supplier<DifferentialDriveWheelSpeeds>`. 

i.e. You’re calling the function and passing the return value rather than passing the function itself

kbyquist, 2022-12-09:
That fixed it, don't know why I didn't try that first. Thanks.