mjansen4857/pathplanner #174: Minimum and ParallelDeadline don't appear in PathPlanner UI

MrRedness, 2022-12-10:
Running PathPlanner-Linux from the latest action on main

mjansen4857, 2022-12-10:
Are you sure you're running the right version? They are definitely there for me, even on the build [here](https://github.com/mjansen4857/pathplanner/actions/runs/3625347787)

MrRedness, 2022-12-10:
It seems something is going on with however options are saved. When I first open an old version of PathPlanner, any new versions I open don't show the new options. When I first open the new version, even the old version shows the new options.

MrRedness, 2022-12-10:
[Screencast from 12-09-2022 09:55:17 PM.webm](https://user-images.githubusercontent.com/16584585/206825408-e7ce1c64-c8fa-47d9-beed-34c44e8d7260.webm)

mjansen4857, 2022-12-10:
So, this happens when you have 2 different builds running at the same time? Seems like that would be some memory management weirdness where the 2 builds may be sharing some memory for some reason. The options in the dropdowns are just all of the options in an enum. So, if 2 versions are sharing that enum then the new version could be missing new options and the old version could have new options appear. Very weird

MrRedness, 2022-12-10:
https://drive.google.com/file/d/1yXhcPJGiST2rIJVkMecqoNILhYFz5xWG/view?usp=sharing

MrRedness, 2022-12-10:
This doesn't seem like a big deal, but it is concerning. I would think Dart programs are well protected from stuff like this.

mjansen4857, 2022-12-10:
This does not happen on windows. I'd assume it is some form of memory optimization (either a flutter thing or a linux thing, though I'd lean towards linux because it doesn't happen on windows) because it sees the 2 different builds as 2 instances of the same program so they can just share any constant memory. They have the same name, same version, etc so I guess it makes sense that it could think its the same program. Not really anything I can do about this, flutter is really high level. If you make a build with a different version number, does it still happen?

MrRedness, 2022-12-13:
Sorry, I'll probably only have time to look at this after (this) finals week. Until then, I'll probably be almost as stressed as a team about to get gated.

mjansen4857, 2022-12-17:
No problem. Honestly, this could probably be closed as it won't actually effect any normal users. They will only have one version installed and there isn't really a need for multiple instances anyways since you can just switch between paths.

Couldn't really do anything about this anyways.

MrRedness, 2022-12-17:
True. I should be able to take a look this week anyway.

mjansen4857, 2023-01-26:
Gonna close this since I can't really do anything about it

closed 2023-01-26 by mjansen4857