mjansen4857/pathplanner #182: PP does not start on Linux

prensing, 2022-12-19:
The latest 2 betas do not start fully on Linux. If I start "./pathplanner" I get a small icon window which has the logo and "PathPlanner" but it then goes no further. No error messages in the terminal. Not sure what else to say.

This is on Linux Mint, latest version. Desktop is Cinnamon. I am happy to try to provide more info, but not sure what you need.

mjansen4857, 2022-12-19:
Your distribution probably has some outdated dependencies compared to the distribution it was built with, which is Ubuntu 22.04. If you build from source it should use whatever version your distribution has.

prensing, 2022-12-19:
I may try, but Linux Mint is actually based on Ubuntu 22.04.

mjansen4857, 2022-12-19:
Trying to run in debug mode at least should give some form of error output if there's some other problem

prensing, 2022-12-19:
Is there a debug mode built into the downloaded jar?  I get no kind of help message, so can't tell.

mjansen4857, 2022-12-19:
You have to build from source to run in debug mode. Readme file has the instructions

mjansen4857, 2022-12-20:
You were probably missing some dependency that got installed while installing flutter

mjansen4857, 2022-12-22:
Going to close this since it was just a missing dependency issue. Not much I can do from my end,

closed 2022-12-22 by mjansen4857