mjansen4857/pathplanner #188: Crash when "Deploy" is clicked while not connected to Robot

itsmichaelyu, 2022-12-27:
The entire program just caches when "Deploy" is clicked, it would be nice to prompt the user with an error instead.

mjansen4857, 2022-12-27:
There already is an error prompt built into flutter but it misses a lot unfortunately. What platform are you running on and where did you install from? It’s likely there’s no permissions to run console commands so that feature should just be disabled for those builds.

itsmichaelyu, 2022-12-27:
I'm on macOS 13.1 x64 with PathPlanner v2023.2.0 from the app store.

mjansen4857, 2022-12-27:
Yeah that shouldn’t even be visible in the App Store build. I’ll look into why that is still there. If you really want to use that feature you’ll need to download the version from GitHub

closed 2022-12-28 by mjansen4857