mjansen4857/pathplanner #195: Path overshoots endpoint in Preview and IRL, but not in Graph

randomstring, 2022-12-31:
I made a simple path that drives forward 2 meters and stops. When I run preview in the GUI the robot clearly overshoots the endpoint and has to backtrack to hit the target endpoint. The robot also does this in real life. Looking at the velocity graph, there is no overshoot.

[image]

[image]

This is using holonomic mode. I can include the path details, but it is just driving straight forward for 2 meters without any rotation, max velocity 2m/s, max acceleration 2m/s^2.

Is this a bug? Why is the back tracking not included in the graph?

mjansen4857, 2022-12-31:
It is in the graph, you just don't notice it because the back tracking you are seeing is still decelerating at -2mps^2.

The backtracking itself is happening because of you control point placement. It should look like this:
[image]

not like this:
[image]

You have the heading of your endpoint set to 180 which is why the robot travels in the 180 direction at the end of the path.

randomstring, 2022-12-31:
Yes. That's exactly what the problem was. I had the final heading set to 180 instead of the desired 0 degrees.

I'm impressed that PathPlanner succeeds at making my impossible request of driving straight to a point from the left, but also arrive from the right come true.

closed 2022-12-31 by randomstring