mjansen4857/pathplanner #196: Non-continuous Rotational Acceleration

randomstring, 2022-12-31:
I made a holonomic test path that drives straight forward 3 meters and rotates 360 degrees. I had to add an intermediate point at the halfway point with a holonomic angle of 180 degrees to force. Looking at the preview it looks good, but looking at the angular velocity graph, it is not smooth. Meaning that the angular acceleration is not continuous. 

The angular velocity for the first half of the of the path is higher than the velocity for the second half. I understand that the angular velocity will be impacted by the speed of the robot, but there shouldn't be a discontinuity at the midpoint. 

Is this just a limitation of how the path planner creates trajectories?

[image]

mjansen4857, 2022-12-31:
Likely another issue with control point placement. If you send the path file I can tell you exactly what is wrong. Path is fine on my end when I make a path according to your description.

randomstring, 2022-12-31:
```json
{
  "waypoints": [
    {
      "anchorPoint": {
        "x": 2.0,
        "y": 3.0
      },
      "prevControl": null,
      "nextControl": {
        "x": 3.0,
        "y": 3.0
      },
      "holonomicAngle": 0.0,
      "isReversal": false,
      "velOverride": null,
      "isLocked": false,
      "isStopPoint": false,
      "stopEvent": {
        "names": [],
        "executionBehavior": "parallel",
        "waitBehavior": "none",
        "waitTime": 0
      }
    },
    {
      "anchorPoint": {
        "x": 3.5,
        "y": 3.0
      },
      "prevControl": {
        "x": 3.4696235779280062,
        "y": 3.0
      },
      "nextControl": {
        "x": 3.5303764220719938,
        "y": 3.0
      },
      "holonomicAngle": 180.0,
      "isReversal": false,
      "velOverride": null,
      "isLocked": false,
      "isStopPoint": false,
      "stopEvent": {
        "names": [],
        "executionBehavior": "parallel",
        "waitBehavior": "none",
        "waitTime": 0
      }
    },
    {
      "anchorPoint": {
        "x": 5.0,
        "y": 3.0
      },
      "prevControl": {
        "x": 4.441675349270015,
        "y": 3.0
      },
      "nextControl": null,
      "holonomicAngle": 360.0,
      "isReversal": false,
      "velOverride": null,
      "isLocked": false,
      "isStopPoint": false,
      "stopEvent": {
        "names": [],
        "executionBehavior": "parallel",
        "waitBehavior": "none",
        "waitTime": 0
      }
    }
  ],
  "maxVelocity": 2.0,
  "maxAcceleration": 2.0,
  "isReversed": null,
  "markers": []
}
```

mjansen4857, 2022-12-31:
Yeah, your midpoint has control points way too close to the anchor point so it is going to do weird things.

Turn this:
[image]

into something like this (since the points completely overlap, you wont be able to drag them since it will just move the entire waypoint. you will have to delete this point and make another):
[image]

When creating paths, you need to make sure that the control points are positioned properly. They aren't just tangent lines, they affect the shape of the path. So, if you have them way too close together, overlapping, etc., it will give you a weird result because it is getting weird input. As a decent rule of thumb I would make sure they never overlap with the anchor point at all.

randomstring, 2022-12-31:
I was able to reduce the non-continuity at the midpoint by adding a velocity override to the midpoint. Using a velocity override lower than the max velocity,  1.8 m/s for a max velocity of 2.0 m/s.

I am wondering if the rotational velocity graph glitches when the angle changes sign. 

[image]

randomstring, 2022-12-31:
I was just looking at the edit data that shows up when you click on the point and ignoring the tangent lines in the GUI. Making the tangent lines as short as I can greatly improved the graph.

[image]

mjansen4857, 2022-12-31:
Generally, making them as short as you can is not a good idea. You should probably create a path and just drag the control points around for a while to see how they affect the path. Moving them to extremes of too close/too far will end up making sharp curves in the path that can cause issues with discontinuity.

randomstring, 2022-12-31:
> They aren't just tangent lines, they affect the shape of the path.

I don't understand how the location of the control points along a straight line with zero curvature can effect the freedom of the robot to rotate to match the desired holomorphic angle. That's ok, I do see the effect when I tinker with the control points. 

I can understand how trying to satisfy robot path heading and the holomorphic heading can be at odds. It makes sense that as you reduce the weight of the path heading (by shortening the length of the control vector) you give more freedom to satisfy the holomorphic heading.

> Generally, making them as short as you can is not a good idea. 

I understand that. Just in this special case, I'm making a straight line path with no curvature at all, so no sharp curves.

Thanks for the feedback. I learned a lot!

closed 2022-12-31 by randomstring